openquantumhardware/qick #5: Count register in qick.averager_program.RAveragerProgram will count ahead of experiment runs

hatlabcz, 2022-01-31:
When running the demo program “02_Sweeping_variables”, I try to change some parameters in the config dict (longer relax_delay, more reps and expts), I found some points in the middle got wrong data.

The config dict I used is:
_config={"res_ch":7, # --Fixed
 "relax_delay":5000, # --Fixed
 "res_phase":0, # --Fixed
 "pulse_style": "const", # --Fixed
 "length":100, # [Clock ticks] 
 "readout_length":200, # [Clock ticks]
 "pulse_gain":0, # [DAC units]
 "pulse_freq": 100, # [MHz]
 "adc_trig_offset": 100, # [Clock ticks]
 "reps":60,  # New variables
 "expts": 40,
 "start":0, # [DAC units]
 "step":50# [DAC units]
 }_

And the result looks like:
[image]

If I change the relax_delay to 0, the result looks normal.
[image]

I try to debug this by looking into the acquire_round function of RAveragerProgram, and I think the problem is probably because the count register (which is accessed via `soc.tproc.single_read(addr=1)` ) is counting ahead of the experiment runs. 

Here I added the following print lines around line 424 of `qick.averager_program` to keep track of this count register.
[image]

And I got outputs like:
[image]

It seems that whenever the tProcessor starts, this count register will directly jump to 8. For example in the result above, the relax_delay is 5ms, but the counter jump to 8 in 3.5e-5s.  (Iterations after this seem to be fine, it will count to 9, 10, 11 after each relax_delay …… )

I think this problem didn't show up for short relax_delay because the time python took to run from reading the counter to acquiring the data is long enough for the board to finish 8  more runs, so we didn't took data that doesn't exist yet. But for long relax_delay , when the counter count to 100, we actually only have 92 data in the buffer.

meeg, 2022-01-31:
Hi, thanks for the detailed report! I was able to reproduce the problem, and it is an issue we know about but haven't made a systematic fix for. At the moment, avoiding this problem requires special care in how you write your programs; that is not explained in the demo notebooks, and we will fix that ASAP.

Yes, you are correct - the counter gets ahead of the experiment. The root cause of this issue is that the core of the tProcessor executes instructions as quickly as possible, and the real-time delays are applied in the timed queues that control the outputs. So if the program pushes some pulses to the timed queues, then does a memory write, the memory write will very often happen before the pulses are played.

This actually affects both AveragerProgram and RAveragerProgram - the problem arises whenever you have a very long delay in your measurement loop, and the effect is that you read stale data from a previous acquisition.

The fix is to add a "waiti" command after the experiment is over, your `body()` would then look like this:
```
    def body(self):        
        self.sync_all(soc.us2cycles(0.05)) # align channels and wait 50ns
        self.trigger_adc(adc1=1, adc2=1,adc_trig_offset=self.cfg["adc_trig_offset"])  # trigger the adc acquisition
        self.pulse(ch=self.cfg["res_ch"], length=self.cfg["length"], play=True) # play readout pulse
        self.sync_all(soc.us2cycles(self.cfg["relax_delay"]))  # sync all channels
        self.waiti(0,0) # make the tProc wait for the sync
        self.synci(1000)
```
With this change, the core of the tProcessor must wait for the timed queues to complete the sync_all before breaking out of the body() and incrementing the counter. If your `relax_delay` is extremely long, it's wasteful to wait until the end of the delay, since you only really need to wait until the data exists in the buffer. So this could be optimized, when it matters.

We're just now in the process of updating the demo notebooks to reflect changes in the library and current understanding of best practices. We will make sure this problem gets fixed in the demos, and we plan to make changes to the library that automatically insert the appropriate wait.

meeg, 2022-02-01:
The demo notebooks no longer suffer from this issue. Generally the fix was as follows:
```
        self.trigger_adc(adc1=1, adc2=1,adc_trig_offset=self.cfg["adc_trig_offset"])  # trigger the adc acquisition
        self.pulse(ch=self.cfg["res_ch"], length=self.cfg["length"], play=True) # play readout pulse
        # control should wait until the readout is over
        self.waiti(0, self.cfg["adc_trig_offset"]+self.cfg["readout_length"]) 
        self.sync_all(soc.us2cycles(self.cfg["relax_delay"]))  # sync all channels
```

I'm going to leave the issue open since we plan to make a more comprehensive fix in the library.

hatlabcz, 2022-02-03:
Hello Sho,

Thank you so much for the detailed explanation and quick fix! That cleared many confusions I used to have about how the asm commands are executed exactly. I've upgraded the code based on your suggestion (basically I just pulled your new update). The counter jump at the beginning is indeed gone! However, it seems that the result I got is still not quite right for long `relax_delay` (the only change I made is setting `"relax_delay": 5000` in the config dict).

Now I get data like:
[image]
while relax_delay=0 gives straight lines.

To check if I'm still getting stale data, I plotted all the data we acquired from the buffer:
```
plt.figure()
plt.plot(prog.di_buf[0])
plt.plot(prog.dq_buf[0])
```

And I got:
[image]
while we know the correct result should just be stairs with no jumping back to 0 in the middle.

The indices of the wrong data points in the middle are 100, 200, 300..., which actually means the first point of each data acquisition cycle (each time count-last_count>=100) is wrong (I'm running at 5ms relax_delay, which is slow enough for each increasement of the counter to be caught in python). I first thought this is again the problem of counting ahead of experiment runs, so I tried to deliberately read data slower than the experiment. I let counter count to 100 but only read the first 90 points in buffer, and then read buffer from address 90 to 180 when counter is already 200. What I found in this test is that now the wrong data shows at 90. So it seems that the first point returned from `soc.get_accumulated` is just always wrong. I'm thinking maybe this is actually rooted in the HDL implementation of the `get_accumulated` functions... but I'm not sure.

Regardless, I try to temporally fix this issue by tweaking the python code. Basically, in each data acquisition cycle, I read data from two address earlier, and read for two extra points, and only keep data starting from index 2.

The codes is: (from line 423  of `averager_program`)
```
        with tqdm(total=total_count, disable=not progress) as pbar:
            tproc.start()
            while count<total_count: # count starts from 0, and count += 1 after each experiment. So in the end count should reach total_count?
                count = tproc.single_read(addr= 1)*readouts_per_experiment
                if count>=min(last_count+100, total_count): # again it should be total_count here? 
                    addr=last_count % soc.get_avg_max_length(0) - 2 # start reading from 2 address earlier.
                    length = count-last_count
                    length -= length%2
                    length_read = length + 2 # read for two more points since the first two will be dropped later

                    for ch in range(2):
                        di,dq = soc.get_accumulated(ch=ch,address=addr, length=length_read)

                        di_buf[ch,last_count:last_count+length]=di[2:length+2] # data after index 2 is the real data we need
                        dq_buf[ch,last_count:last_count+length]=dq[2:length+2]

                    last_count+=length
                    self.stats.append( (time.time(), count,addr, length))
                    pbar.update(last_count-pbar.n)
```
In the above code, I also adjusted the stop condition for the while loop and counting, since I found after I fixed the points in the middle, the last point is still always wrong (the original condition seems to count one less point).

Using the code above, I was able to get correct results that finally looks like:
[image]
[image]

I tested this with some different conditions in config dict. And it seems all work fine. However, a better solution to this might need a more careful look at the `get_accumulated` function...

Btw.....this might be a dumb question: when you say that the instructions in main processor of tProc is executed as quick as possible, do you mean one asm command is executed per clock cycle, or is there any kind of parallelization in it? 

Thank you again for your help!

Chao

meeg, 2022-02-03:
Thanks, this is pretty convincing - it does look like there's a bug either in the HDL or in the Python code that does the DMA transfer. I will look at this on Monday (have some other projects I'm busy with this week).

Thanks again for catching this and investigating thoroughly.

About the tProc: it's a pretty simple processor, there's no parallelization or pipelining so one asm command is executed at a time. Also, some instructions take multiple clock cycles to execute. Since the timed queues and other peripherals operate in parallel with the main processor, this is generally OK.

meeg, 2022-02-22:
Hi, sorry for the delay - I was able to reproduce what you saw. It is probably an HDL bug, and @leoeltipo is looking at it. Meanwhile I incorporated your workaround in the get_accumulated and get_decimated methods, so now everything seems OK without making changes to averager_program.py. I think I also fixed the loop boundary bug that you found.

hatlabcz, 2022-02-22:
That sounds great. Thank you! And thank you again for explaining how the asm commands are executed.

meeg, 2022-04-01:
> We're just now in the process of updating the demo notebooks to reflect changes in the library and current understanding of best practices. We will make sure this problem gets fixed in the demos, and we plan to make changes to the library that automatically insert the appropriate wait.

Long overdue, but with the latest PR (#30) there are now two ways to do this:
* wait_all() is a "smart waiti" that automatically inserts the correct wait - analogous to sync_all()
* measure() wraps pulse(), trigger(), and (if you set the correct options) wait_all and sync_all.

The send_receive_pulse demo has some more information, and most of the demo notebooks now use measure().

closed 2022-04-01 by meeg