openquantumhardware/qick #31: first points read from buffer are always the first points of the buffer

meeg, 2022-04-01:
Hello Sho,

Thank you so much for the detailed explanation and quick fix! That cleared many confusions I used to have about how the asm commands are executed exactly. I've upgraded the code based on your suggestion (basically I just pulled your new update). The counter jump at the beginning is indeed gone! However, it seems that the result I got is still not quite right for long `relax_delay` (the only change I made is setting `"relax_delay": 5000` in the config dict).

Now I get data like:
[image]
while relax_delay=0 gives straight lines.

To check if I'm still getting stale data, I plotted all the data we acquired from the buffer:
```
plt.figure()
plt.plot(prog.di_buf[0])
plt.plot(prog.dq_buf[0])
```

And I got:
[image]
while we know the correct result should just be stairs with no jumping back to 0 in the middle.

The indices of the wrong data points in the middle are 100, 200, 300..., which actually means the first point of each data acquisition cycle (each time count-last_count>=100) is wrong (I'm running at 5ms relax_delay, which is slow enough for each increasement of the counter to be caught in python). I first thought this is again the problem of counting ahead of experiment runs, so I tried to deliberately read data slower than the experiment. I let counter count to 100 but only read the first 90 points in buffer, and then read buffer from address 90 to 180 when counter is already 200. What I found in this test is that now the wrong data shows at 90. So it seems that the first point returned from `soc.get_accumulated` is just always wrong. I'm thinking maybe this is actually rooted in the HDL implementation of the `get_accumulated` functions... but I'm not sure.

Regardless, I try to temporally fix this issue by tweaking the python code. Basically, in each data acquisition cycle, I read data from two address earlier, and read for two extra points, and only keep data starting from index 2.

The codes is: (from line 423  of `averager_program`)
```
        with tqdm(total=total_count, disable=not progress) as pbar:
            tproc.start()
            while count<total_count: # count starts from 0, and count += 1 after each experiment. So in the end count should reach total_count?
                count = tproc.single_read(addr= 1)*readouts_per_experiment
                if count>=min(last_count+100, total_count): # again it should be total_count here? 
                    addr=last_count % soc.get_avg_max_length(0) - 2 # start reading from 2 address earlier.
                    length = count-last_count
                    length -= length%2
                    length_read = length + 2 # read for two more points since the first two will be dropped later

                    for ch in range(2):
                        di,dq = soc.get_accumulated(ch=ch,address=addr, length=length_read)

                        di_buf[ch,last_count:last_count+length]=di[2:length+2] # data after index 2 is the real data we need
                        dq_buf[ch,last_count:last_count+length]=dq[2:length+2]

                    last_count+=length
                    self.stats.append( (time.time(), count,addr, length))
                    pbar.update(last_count-pbar.n)
```
In the above code, I also adjusted the stop condition for the while loop and counting, since I found after I fixed the points in the middle, the last point is still always wrong (the original condition seems to count one less point).

Using the code above, I was able to get correct results that finally looks like:
[image]
[image]

I tested this with some different conditions in config dict. And it seems all work fine. However, a better solution to this might need a more careful look at the `get_accumulated` function...

Btw.....this might be a dumb question: when you say that the instructions in main processor of tProc is executed as quick as possible, do you mean one asm command is executed per clock cycle, or is there any kind of parallelization in it? 

Thank you again for your help!

Chao

_Originally posted by @hatlabcz in https://github.com/openquantumhardware/qick/issues/5#issuecomment-1029071736_

meeg, 2022-04-01:
As noted in #5, the software works around this bug so users should be totally safe. But there's still a firmware bug which should be fixed (at low priority).

dmartin2-fnal, 2025-04-16:
@meeg is it worth it to invest some time in looking into this bug in the firmware?

meeg, 2025-04-16:
Not at this point - the workaround is fine, and if #327 gets implemented we will no longer be using this buffer at all.