openquantumhardware/qick #51: RFQickSoc() usage; missing SPI IP files

mutasemodeh, 2022-06-29:
Greetings from UC Berkeley. 

Thanks for the phenomenal work! I have followed the tutorials and got myself familiar with qick library.

I am now planning to perform an upconverting with an external LO and I have purchased LMX2595 evaluation board to do that. It is similar to the one used in the RFboard front end. I would like to be able to control it via SPI through Qick. I have followed the schematic and connected the SPI wires accordingly. However the SPI IPs are missing and I can't Initialize the board using **RFQickSoc()**. Any suggestion on how to load them and set the LO frequency to a particular frequency? 

Thanks,
Mutasem

[image]

[image]

mutasemodeh, 2022-06-29:
Thank you Sho for the quick response. That is correct, TICS software works fine and unfortunately it is Windows-only. I am trying to give autonomy to Xilinx board so I can tune the LO in a python slow soft loop.

- Thanks for sharing the bitfile, I have managed to install it and the **RFQickSoc()** works well now. 
- The LMX2595 LO is not responding nonetheless. After digging into the the rfboard.py library, it seems that the scripted registers are meant for **ADF4372** and it is confirmed by the data sheet. Is there a newer version of rfboard.py that was not pushed to the repository yet?
- Yes, I am planning to take the reference input from ZCU111, but I have yet to figure out how to generate a clean 100MHz or similar and share it with LMX2595EVM. 

[image]

meeg, 2022-06-29:
Got it, makes sense.

You're right - sorry, I am not so in touch with the history of the RF board design. It turns out that the RF board schematic in the repository is for a newer version of the board (v2) than what rfboard.py was written for (v1). The v1 schematic is now added to the repo (#52). I'm sorry if this meant you bought the wrong PLL!

Since we don't have the v2 board yet, there is probably no Python code for configuring the LMX2595 (I'll ask) - you will need to write your own using the TICS-generated register map.

RFQickSoc configures the ZCU111's onboard clocks to output 122.88 MHz on J108. That's the reference clock we use for the RF board, and it should work for you.

mutasemodeh, 2022-08-08:
Thanks, meeg, I've finally managed to find sometime for it and get it to work. Below is a little bit of details for those who are interested in assembling and programming the LMX2595 evaluation board. 

I used x2 LMX2595EVM oscillators and x2 SN74AXC4T774EVM logic level transelator for SPI programming via QICK. The rest are a bunch of regulators, filters, mixers, and amplifiers for my experiment. 

[image]

I haven't got the time to program functions for LMX2595. For now, I generated the register values via TICS PRO, saved them to a text file, and asked QICK to send them via its SPI interface. Below is a simple example to set one of the LO at ~8GHz 

```
# Import the QICK drivers and auxiliary libraries
from qick import *
from qick.rfboard import RFQickSoc
from numpy import loadtxt
homeDir="/home/xilinx/jupyter_notebooks/qick/qick_lib/qick/"

# Load bitstream with custom overlay
# soc = QickSoc()
soc = RFQickSoc(bitfile="/home/xilinx/jupyter_notebooks/qick/qick_lib/qick/qick_111_rfbrd.bit")

soccfg = soc
soc.lo_spi.config(lsb="msb")
# SPI.
nch=2 # number of LO connected
le0=[0] # lo0 connected to port 0
le1=[1] # lo1 connected to port 1
en_l="low" # enable level is low for lmx2595
# enable bit for lo1
ch0_en = soc.lo_spi.en_level(nch, le0, en_l)
print(ch0_en)
# enable bit for lo2
ch1_en = soc.lo_spi.en_level(nch, le1, en_l)
print(ch1_en)
# All CS to high value.
soc.lo_spi.SPI_SSR = 0xff

regs = loadtxt(homeDir+"HexRegisterValues-Copy1.txt",dtype='str', comments="#", delimiter="\t", unpack=False,usecols=1)
data=bytes.fromhex('000020')
# rec=soc.lo_spi.send_receive_m(data,ch0_en,"")
data=bytes.fromhex('000000')
rec=soc.lo_spi.send_receive_m(data,ch0_en,"")
for i in range(len(regs)):
    hexdata=regs[i][2:]
    data=bytes.fromhex(hexdata)
    rec=soc.lo_spi.send_receive_m(data,ch0_en,"")

```

[image]

I hope that is helpful and thanks again for the amazing project!

meeg, 2022-08-08:
That's really great to hear, and a nice setup! Thanks for the detail, this will absolutely be helpful. Sorry again for the mixup with the schematics.

closed 2022-08-08 by meeg

arka-firmware01, 2025-04-11:
[https://github.com/openquantumhardware/qick/issues/51#issuecomment-1207739016](url)

**The front end with the SPI interface TI LMX2595 is already set up. We'd like to understand how the backend using the ZCU111 was configured, particularly the environment setup for the ADC and DAC interfaces. How was this integration achieved in your case?**

arka-firmware01, 2025-04-11:
> Thanks, meeg, I've finally managed to find sometime for it and get it to work. Below is a little bit of details for those who are interested in assembling and programming the LMX2595 evaluation [...]