openquantumhardware/qick #61: Streaming Samples through the QICK Model to/from the PS

jramsey123, 2022-08-01:
I'd like the option of streaming received samples  from the ADC to the PS over DMA, and vice versa streaming samples from the PS to the DAC over DMA.  The QICK examples I've seen have a finite number of samples processed (in or out) as a number of clock ticks.  Is there a QICK demo that uses the tProc as a streaming controller, so to speak?  And if there isn't, do you see any difficulty in achieving this?  Thank you.

meeg, 2022-08-01:
I suspect that what you want is pretty different from what QICK is designed to do - QICK only plays pulse envelopes that are loaded prior to program start, and only outputs ADC data after downconversion+decimation and boxcar averaging - but can you give a more specific use case?

stoughto, 2022-08-01:
It is possible that the functionality you want is in firmware we are
developing for MKIDs detectors.  Please describe the use case and we can
let you know.

jramsey123, 2022-08-02:
Our initial application is a RADAR test system that, e.g., detects and simulates Doppler effect in a smaller platform based on the ZCU208.  Initially we want to relay raw IQ, then follow that will IQ processing down on the FPGA.  We can control the data rates to/from the PS by adjusting the sampling rate initially, and later with FPGA processing the sampling rates can be increased.  The MKID work done at UCSB that was presented at IAC is more processing than we need initially, and we don't need to channelize the received RADAR data.  I knew QICK was a short signal burst implementation given their requirements, but I was hoping to leverage it to add the streaming capability then extend the block design with IQ processing later on.  Thank you for your response.  Maybe your MKID work would help?

meeg, 2025-03-05:
Cleaning up old issues; feel free to reopen if this is still of interest (sorry for dropping the ball on it).

closed 2025-03-05 by meeg