openquantumhardware/qick #70: R-type set command that takes a single register

jacobfeder, 2022-09-18:
The assembly "set" command takes in a bunch of registers for controlling various pulse parameters specific to the DAC (freq, phase, gain, etc.). The documentation shows:
set ch, p, $ra, $rb, $rc, $rd, $re, $rt:

This makes it quite annoying to use the set instruction for ADC / digital marker triggering. E.g. if I want to trigger an ADC readout or digital marker at a time defined in a register (so seti won't do), I need to configure $ra, $rb, $rc, $rd, $re, $rt instead of just $rt and a single register containing all of the marker/trigger bits.

I think it would be nice to have a separate instruction for this purpose e.g.
set2 ch, p, $ra, $rt
where $ra contains all of the marker/trigger bits.

meeg, 2022-09-18:
For command arguments where the value doesn't matter (and in this case it really doesn't, the higher bits of that tProc port go nowhere), you can use whatever registers you want, regardless of whether you've written values into them (because the value doesn't matter!). Readability is probably improved by using $0.

So I think `set ch, p, $ra, $0, $0, $0, $0, $rt` is what you want - will that do? If you really don't want those $0s to appear in your code, you/I can add a wrapper like this to QickProgram (I haven't tested this):
```
def seta(self, rp, reg, comment=None):
    self.set(rp, reg, 0, 0, 0, 0, comment)
```

Either way, I don't think an actual change to the instruction set is needed. But please correct me if I'm missing something.

jacobfeder, 2022-09-18:
Is it documented somewhere how the registers get mapped to the tProc port? It wasn't really clear from
https://github.com/openquantumhardware/qick/blob/main/firmware/tProcessor_64_and_Signal_Generator_V4.pdf
Another way of asking the same question: does the entire $ra register (all 64 bits) get sent to the tProc port? I assumed the lower bits of each of $ra/b/c/e fed into some subset of the tProc output port.

meeg, 2022-09-18:
Registers are 32 bits, and `set` pushes the complete contents of each register to the tProc port. 5 registers of 32 bits each = 160 bits.

jacobfeder, 2022-09-18:
Ah ok. So for the ADC/marker register, the remaining 4*32=128 bits are unused?

meeg, 2022-09-18:
Correct - otherwise seti (which can only output one 32-bit value) wouldn't be able to drive the marker/trigger output. Actually, no firmware in current use wires anything outside the first 16 bits of the marker/trigger output, but there is no deep reason for that.

jacobfeder, 2022-09-18:
Ok got it. I will use `set ch, p, $ra, $0, $0, $0, $0, $rt`. Thanks!! Closing

closed 2022-09-18 by jacobfeder