openquantumhardware/qick #73: ZCU216 Board Description

EeshGupta, 2022-09-28:
When I  try to print out the following:
`soc = Pyro4.Proxy(ns.lookup(server_name))
soccfg = QickConfig(soc.get_cfg())
print(soccfg)`

The print line produces the following error: 
`---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-18-e1f448585823> in <module>
      1 soc = Pyro4.Proxy(ns.lookup(server_name))
      2 soccfg = QickConfig(soc.get_cfg())
----> 3 print(soccfg)

c:\_lib\python\qick\qick_lib\qick\qick_asm.py in __str__(self)
     31 
     32     def __str__(self):
---> 33         return self.description()
     34 
     35     def __getitem__(self, key):

c:\_lib\python\qick\qick_lib\qick\qick_asm.py in description(self)
     76                 readout['avg_maxlen'], readout['buf_maxlen'], readout['trigger_bit'], readout['tproc_ch']))
     77 
---> 78         lines.append("\n\t%d DACs:" % (len(self['dacs'])))
     79         for dac in self['dacs']:
     80             tile, block = [int(c) for c in dac]

c:\_lib\python\qick\qick_lib\qick\qick_asm.py in __getitem__(self, key)
     34 
     35     def __getitem__(self, key):
---> 36         return self._cfg[key]
     37 
     38     def __setitem__(self, key, val):

KeyError: 'dacs'`

Indeed when I print out the configuration using 
`soccfg.get_cfg()`
I get the the output below which does not contain 'dacs'
`{'board': 'ZCU216',
 'refclk_freq': 245.76,
 'fs_proc': 430.08,
 'gens': [{'type': 'axis_signal_gen_v4',
   'maxlen': 65536,
   'b_dds': 32,
   'switch_ch': 0,
   'tproc_ch': 1,
   'dac': '20',
   'fs': 6881.280000000001,
   'f_fabric': 430.08,
   'samps_per_clk': 16,
   'maxv': 32766,
   'maxv_scale': 1.0},
  {'type': 'axis_signal_gen_v4',
   'maxlen': 65536,
   'b_dds': 32,
   'switch_ch': 1,
   'tproc_ch': 2,
   'dac': '21',
   'fs': 6881.280000000001,
   'f_fabric': 430.08,
   'samps_per_clk': 16,
   'maxv': 32766,
   'maxv_scale': 1.0},
  {'type': 'axis_signal_gen_v4',
   'maxlen': 65536,
   'b_dds': 32,
   'switch_ch': 2,
   'tproc_ch': 3,
   'dac': '22',
   'fs': 6881.280000000001,
   'f_fabric': 430.08,
   'samps_per_clk': 16,
   'maxv': 32766,
   'maxv_scale': 1.0},
  {'type': 'axis_signal_gen_v4',
   'maxlen': 65536,
   'b_dds': 32,
   'switch_ch': 3,
   'tproc_ch': 4,
   'dac': '23',
   'fs': 6881.280000000001,
   'f_fabric': 430.08,
   'samps_per_clk': 16,
   'maxv': 32766,
   'maxv_scale': 1.0},
  {'type': 'axis_signal_gen_v4',
   'maxlen': 65536,
   'b_dds': 32,
   'switch_ch': 4,
   'tproc_ch': 5,
   'dac': '30',
   'fs': 6881.280000000001,
   'f_fabric': 430.08,
   'samps_per_clk': 16,
   'maxv': 32766,
   'maxv_scale': 1.0},
  {'type': 'axis_signal_gen_v4',
   'maxlen': 65536,
   'b_dds': 32,
   'switch_ch': 5,
   'tproc_ch': 6,
   'dac': '31',
   'fs': 6881.280000000001,
   'f_fabric': 430.08,
   'samps_per_clk': 16,
   'maxv': 32766,
   'maxv_scale': 1.0},
  {'type': 'axis_signal_gen_v4',
   'maxlen': 65536,
   'b_dds': 32,
   'switch_ch': 6,
   'tproc_ch': 7,
   'dac': '32',
   'fs': 6881.280000000001,
   'f_fabric': 430.08,
   'samps_per_clk': 16,
   'maxv': 32766,
   'maxv_scale': 1.0}],
 'readouts': [{'avg_maxlen': 16384,
   'buf_maxlen': 1024,
   'b_dds': 32,
   'adc': '20',
   'fs': 2457.6,
   'f_fabric': 307.2,
   'trigger_bit': 14,
   'tproc_ch': 0},
  {'avg_maxlen': 16384,
   'buf_maxlen': 1024,
   'b_dds': 32,
   'adc': '22',
   'fs': 2457.6,
   'f_fabric': 307.2,
   'trigger_bit': 15,
   'tproc_ch': 1}],
 'iqs': [],
 'tprocs': [{'trig_output': 0, 'pmem_size': 8192.0, 'dmem_size': 4096}]}`

meeg, 2022-09-28:
I suspect you might be running different versions of the QICK library on the PC and on the QICK - specifically, an old version on the QICK.

Could you double-check the versions? One way (depending on how you installed the library) would be to run `import qick; print(qick)` to find what directory you're loading the library from, then `git log` in that directory.

EeshGupta, 2022-09-28:
looks up to date

meeg, 2022-09-28:
Yes, that's up to date - this is the version on the QICK, or on the PC? (I am assuming you are running a Pyro server on the QICK, and trying to connect to that from a PC.)