openquantumhardware/qick #78: Negative differential DAC output from QICK signal generator

mutasemodeh, 2022-10-03:
Hi, 

I am looking into generating a variable +ve or -ve DC signal using one of the QICK signal generators. To do so, I have connected the differential outputs of DAC228_T0_CH0 P and N terminals to an operational amplifier with a split-power supply (+3.3V, -3.3V) hoping to reject the common mode and generate a +ve/-ve DC signal. 

```
self.declare_gen(ch=0, nqz=1)
self.set_pulse_registers(ch=0, freq=0, length=1000, phase=0, gain=g, style="const") 
```
 However, the DAC P voltage is always equal to or larger than the DAC N voltage when sweeping the gain value. I was expecting the signal generator to behave  roughly like this with a bit of offset:
  
  DAC P=DAC N, when gain=0,
  DAC P>DAC N, when gain>0,
  DAC P< DAC N when gain<0.
  
Is there a way to control this?

Thanks,
Mutasem

meeg, 2022-10-03:
That's odd - this makes sense and should work (outsel="input" is another way to get DC outputs, but only works with arb style). I don't have a ZCU111 on my desk right now but the ZCU216 should work the same - here is a scope shot where ch 2 = P, ch 3 = N on DAC 2_231, I am running the following trivial program similar to yours:

```
class DummyProgram(AveragerProgram):
    def initialize(self):
        self.declare_gen(ch=4, nqz=1)
        self.set_pulse_registers(ch=4, freq=0, length=100, phase=0, gain=-30000, style="const")
        self.pulse(ch=4,t=0)

config = {
    'reps': 1
}
prog = DummyProgram(soccfg, config)
iq_list = prog.acquire_decimated(soc, progress=True)
```

[image]

If I set gain = 30000, the traces flip as they should.

Maybe something is wrong with your amplifier circuit?

mutasemodeh, 2022-10-06:
Thanks, Meeg, I realized my amplifier has an unbalanced input impedance, causing extra reflection on one of the ports. Thanks for taking the time to test it.

Mutasem

closed 2023-03-24 by meeg