openquantumhardware/qick #82: Mapping between DAC pulse_gain and actual AXI number that feed into the DAC?

zhizhenzhong, 2022-10-08:
Hi folks,

I have a question about the mapping between DAC `pulse_gain` in the QICK configuration setting (in instructions it suggest us to try pulse_gain from 500 to 30000 DAC units). I want to know how does these numbers mapped to the AXI input number of the 14-bit DAC (0-8191)? Same question for the 12-bit ADC - how does the `acquire_decimated` return value mapped to the actual ADC output AXI word?

Thanks!
Zhizhen

meeg, 2022-10-08:
I am not sure how deep an answer you want here. I would suggest doing some sanity checks on a real board using the demos as a starting point. If you really want to pick things apart on a deep level, you might look at the firmware source (the DAC math is basically all in https://github.com/openquantumhardware/qick/blob/main/firmware/ip/axis_signal_gen_v6/src/signal_gen.v, for the ADC math you will have to look in a few files).

The first basic point is that while the AXI interfaces to the Xilinx DAC+ADC use unsigned values like you say, all user-facing numbers in QICK (gains, envelopes, decimated+accumulated readout) are signed. So zero volts (or P=N, in terms of the differential ports) is 0, not 4096 or 1024 or whatever.

The gains on the DAC side are bit-shifted such that the max gain and envelope (both are signed 16-bit ints) will not result in overflow at the DAC.
* For direct output (without DDS upconversion) a gain of 2^15-1 and a (real) envelope of 2^15-1 maps to full-scale DAC output.
* Similarly, for DDS output (without envelope) a gain of 2^15-1 maps to full-scale DAC output.
* For upconverted output (the default, with DDS upconversion) the envelope is complex, which adds a factor of sqrt(2) to the max magnitude. So a gain of 2^15-1 and a purely real or purely imaginary envelope of 2^15-1 maps to half-scale DAC output; you can get to 1/sqrt(2) full-scale by setting I=Q=2^15-1.

On the ADC side the math is more complicated, since there is the decimating filter. I can say that full scale on the ADC will not saturate the range (signed 16-bit) of the decimated value.

meeg, 2023-03-24:
I don't know if your question was addressed, but please feel free to reopen if you want to continue this topic.

closed 2023-03-24 by meeg