openquantumhardware/qick #89: Readout with the board RFSoC4x2

JavierSerranoGarcia, 2022-10-21:
I am trying to characterize a qubit using the notebook for the qubit demo. The demo suggests doing the readout with a mixer and LO (on a ZCU111) but considering that the RFSoc4x2 board has a sampling frequency of 10Gsps I am trying to do it without mixer. But I am not getting any nice results.

Is it possible to do readouts on this board without mixers? or should I use mixer and LO?

meeg, 2022-10-21:
Yes, the RFSoC4x2 and ZCU216 are both suitable for qubit readout without mixers, and people have had good results. If you can share more details on your setup and results, I'm happy to give some thoughts.

meeg, 2022-10-26:
The amplitude seems very small. A few ideas:
* Your `adc_trig_offset` parameter, at 1000 clock ticks, is oddly large and round. How did you arrive at this value? I don't have the 4x2 firmware's clock frequencies on hand, but I think you will find that `soccfg.clocks2us(1000)` tells you that is tens of us. Do you really have that much cable delay from your DAC to your ADC? If you don't, you are probably acquiring data only after the end of your readout pulse. The adc_trig_offset parameter should be set by looking at decimated waveforms and identifying the parameter value that puts the start of your readout pulse at (or slightly before) your readout window - what the demo notebook calls a TOF measurement. The goal is to only acquire data while the readout pulse is at the ADC, not before or after.
* If you can't see the pulse in the TOF measurement, you could try a direct loopback from the DAC to the ADC - make sure the cabling to the fridge is not the problem and that you have the correct channel numbers.
* Maybe you need more amplification to boost the signal going into the ADC?

JavierSerranoGarcia, 2022-10-31:
Thank you so much for your advice. Now I can do the resonator and qubit spectroscopy and the Rabi oscillation

meeg, 2022-11-02:
Good to hear!

closed 2022-11-02 by meeg