openquantumhardware/qick #93: Problem with  Pynq 3.0 on RFSoc4x2 + Qick

JavierSerranoGarcia, 2022-11-02:
I have just downloaded the new Pynq 3.0 and booted the RFSox4x2 board and after installing Qick running on Jupyter

```
%pylab inline
from qick import *
from tqdm.notebook import tqdm
soc = QickSoc()
soccfg = soc
```

I got this message
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Input In [4], in <cell line: 2>()
      1 # Load bitstream with custom overlay
----> 2 soc = QickSoc()
      3 # Since we're running locally on the QICK, we don't need a separate QickConfig object.
      4 # If running remotely, you could generate a QickConfig from the QickSoc:
      5 #     soccfg = QickConfig(soc.get_cfg())
   (...)
      8 #         f.write(soc.dump_cfg())
      9 #     soccfg = QickConfig("qick_config.json")
     10 soccfg = soc

File /home/xilinx/jupyter_notebooks/qick/qick_lib/qick/qick.py:1734, in QickSoc.__init__(self, bitfile, force_init_clks, ignore_version, no_tproc, **kwargs)
   1732 self._tproc = self.axis_tproc64x32_x8_0
   1733 self._tproc.configure(self.axi_bram_ctrl_0, self.axi_dma_tproc)
-> 1734 self['fs_proc'] = get_fclk(self.parser, self.tproc.fullpath, "aclk")
   1736 self.map_signal_paths()
   1738 self._streamer = DataStreamer(self)

File /home/xilinx/jupyter_notebooks/qick/qick_lib/qick/helpers.py:157, in get_fclk(parser, blockname, portname)
    143 """
    144 Find the frequency of a clock port.
    145 
   (...)
    153 :rtype: float
    154 """
    155 xmlpath = "./MODULES/MODULE[@FULLNAME='/{0}']/PORTS/PORT[@NAME='{1}']".format(
    156     blockname, portname)
--> 157 port = parser.root.find(xmlpath)
    158 return float(port.get('CLKFREQUENCY'))/1e6

AttributeError: 'RuntimeMetadataParser' object has no attribute 'root'
```
Any idea about the issue?

(With the old version I have no issue)

meeg, 2022-11-02:
Pynq 3.0 changes parts of the "parser" API that we use. It's probably a simple fix to get things working on 3.0, but we have not had time to try running on 3.0.

For now, I think you should use the Pynq 2.7 image unless you have a specific need for 3.0.

JavierSerranoGarcia, 2022-11-09:
Ok, thank you

closed 2022-11-09 by JavierSerranoGarcia