openquantumhardware/qick #95: Changing tProc clk frequency (zcu111 board)

yyjun22, 2022-11-09:
Hi i'm using zcu111 board to make DC signal and RF signal

I want to know how to change clk frequency of tProc which is now 384MHz.

Do I have to change some firmware code files? or are there any simple ways to change tProc clk frequency?

thank you

meeg, 2022-11-09:
Why do you want to do this? What tProc frequency do you want?

It's not simple - you would need to recompile the firmware from source (you will find basic instructions in `firmware/README.md`), and then you will need to change the clock that goes into the tProc. It's not a huge deal but takes some experience with firmware work.

meeg, 2023-03-24:
Please feel free to reopen if you want to continue this topic.

closed 2023-03-24 by meeg