openquantumhardware/qick #96: Bug in tproc

EeshGupta, 2022-11-11:
Hi I am getting a weird bug when I do start up qick via a client 
Input (connecting to nameserver and printing out qick soc object): 
>import Pyro4
from qick import *
from qick import QickConfig
Pyro4.config.SERIALIZER = "pickle"
Pyro4.config.PICKLE_PROTOCOL_VERSION=4

>ns_host = '192.168.0.133'  #ip of client pc where the nameserver is running
ns_port = 8000
server_name = "myqick"

>ns = Pyro4.locateNS(host=ns_host, port=ns_port)

> soc = Pyro4.Proxy(ns.lookup(server_name))
soccfg = QickConfig(soc.get_cfg())
print(soccfg)  

Output:

> KeyError                                  Traceback (most recent call last)
Cell In [8], line 3
      1 soc = Pyro4.Proxy(ns.lookup(server_name))
      2 soccfg = QickConfig(soc.get_cfg())
----> 3 print(soccfg)
File c:\users\chakram_meas\documents\github\qick\qick_lib\qick\qick_asm.py:48, in QickConfig.__str__(self)
     47 def __str__(self):
---> 48     return self.description()
File c:\users\chakram_meas\documents\github\qick\qick_lib\qick\qick_asm.py:88, in QickConfig.description(self)
     86 lines.append("\n\t%d readout channels:" % (len(self['readouts'])))
     87 for iReadout, readout in enumerate(self['readouts']):
---> 88     if readout['tproc_ctrl'] is None:
     89         lines.append("\t%d:\t%s - controlled by PYNQ" % (iReadout, readout['ro_type']))
     90     else:
KeyError: 'tproc_ctrl'

This error does not occur when I run qick directly from xilinx board (instead of client/external pc)

EeshGupta, 2022-11-11:
my qick version is up to date (0.2); I installed it a day ago (Nov 10)

meeg, 2022-11-11:
Thanks for the info - did you install the latest qick version on both the board and the PC? I would expect this error if you have an out-of-date version on the board.

EeshGupta, 2022-11-11:
Yes I cloned it today itself

meeg, 2022-11-11:
OK, thanks for checking. Some more checks:

* On the PC, can you show me the output of `print(soc.get_cfg())`?
* On both the board and the PC, can you show me the output of `print(qick)`? I want to double-check that your up-to-date clone is what is actually being run.
* On both the board and the PC, can you run `from qick import QickProgram`, `help(QickProgram.ch_page_ro)`? That is a method that is defined in the new version (which I think is what you're running on the PC) but not the old version (what I think you have on the board).

EeshGupta, 2022-11-14:
Hi @meeg , so I restarted the PC and now I get the expected ouput. I think in updating the qick installed on my PC, I had to restart  for those changes to be fully installed(?).
Thanks for the prompt response and the help!

closed 2022-11-14 by EeshGupta