openquantumhardware/qick #102: Changing signal generator

Cameron-Spence-Riverlane, 2022-12-15:
Hi, thank you for this excellent softare. I have a (potentially dumb) question on the signal generators - how is it possible to switch the type of generator? I am working on ZCU216 and all generators are axis_signal_gen_v4; I would like to use axis_sg_mux4_v2. Does it require recompiling the bitfile to change? Is there any documentation that would help (besides the qick docs)?

Cheers

meeg, 2022-12-15:
Yes, it requires recompiling the bitfile. There are some instructions on setting up the firmware project in https://github.com/openquantumhardware/qick/tree/main/firmware; you would then be able to swap out the generators as desired.

That said, you are welcome to try https://www.slac.stanford.edu/~meeg/qick/fw/2022-12-07_q3diamond_cdcsync/ which has (among things) an axis_sg_mux4_v2 generator.

Cameron-Spence-Riverlane, 2022-12-15:
Very helpful, thanks!

closed 2022-12-15 by Cameron-Spence-Riverlane