openquantumhardware/qick #103: Incorrect register setting for muxed gen

Cameron-Spence-Riverlane, 2023-01-06:
Hi,
I have been having difficulty getting a continuous (multiple) tone output from a mux channel. in `MultiplexedGenManager.write_regs` there is `
        if 'length' in params:
            self.set_reg('freq', params['length'], defaults=defaults)`
Which seems incorrect to me, the length parameter is being passed to the frequency register. Is this a deprecated method?
Could you point me to working code that does something similar?

Thank you kindly

meeg, 2023-01-06:
Sorry for the confusion - this is hacky but correct code, and `MultiplexedGenManager.write_regs` is being used under the hood by anyone using a muxed gen. (I assume you're not calling it by hand, only through `QickProgram.set_pulse_registers`, right?)

The different generators have different register maps, but the software always defines the same set of register names: https://github.com/openquantumhardware/qick/blob/3a152dbe4d5afa74d59809a034eeebdc7f7f9472/qick_lib/qick/qick_asm.py#L1033 Those names were chosen to make sense with the default "full-speed" generator, but are really just arbitrary keys that map to register numbers.

The muxed gen uses only two registers, for the length and mask. There's no "length" or "mask" in the list of names, so we use "freq" and "phase" instead. But that's OK.

That said - obviously there is some reason your test isn't working. If you'd like to share your program (preferably enough of the Python notebook/code that I can run it myself, but the ASM you get from `print(prog)` would be good enough) I can take a look.

Cameron-Spence-Riverlane, 2023-01-09:
Hi, thank you for the clarification. In that case I am unsure why my code fails. My asm output is
 `
  regwi 3, $22, 1000;                           //freq = 1000
  regwi 3, $23, 7;                              //mask = [0, 1, 2]
  regwi 3, $27, 0;                              //t = 0
  set 6, 3, $22, $23, $0, $0, $0, $27;          //ch = 6, pulse @t = $27
  synci 200;
  end ;`

which seems OK to me. I am trying a simple script:
`
from qick import QickSoc, QickProgram
soc = QickSoc()
prog = QickProgram(soc)
prog.declare_gen(6,nqz=1,mixer_freq=0,mux_freqs=[400,500,600,700],mux_gains=[1.0,1.0,1.0,1.0])
prog.config_gens(soc)
prog.set_pulse_registers(6,style="const",length=1000,mask=[0,1,2])
prog.pulse(6)
prog.synci(200)
prog.end()
prog.load_program(soc)
soc.tproc.start()`

I just have the output connected to an oscilloscope as a test, and I get no response.
Perhaps I have missed something with setting up the gen channel?

Thanks for your help!

Cameron-Spence-Riverlane, 2023-01-09:
@meeg I managed to solve the problem, the issue was with using length=1000 in set_pulse_registers. When I set it to 0 it works ok, although it does not run continuously. "periodic" mode does not work for the mux gen, so is there a way I can run a 3 tone output indefinitely? 

I am still unsure on the use of "length" in set_pulse_registers - is this setting the mixer frequency?

Cameron-Spence-Riverlane, 2023-01-10:
I see, thank you for the clarification. I am not sure why I did not see the same pulse using length=1000, it should have still triggered on the scope.

closed 2023-01-10 by Cameron-Spence-Riverlane