openquantumhardware/qick #105: RuntimeError: Frequency 245.76 MHz is not valid.

zhizhenzhong, 2023-01-25:
Hi, I am using the ZCU216-PYNQ2.7 image (https://github.com/sarafs1926/ZCU216-PYNQ/issues/1) for my ZCU216 board. I am using a ZCU216 overlay bitstream (https://slac.stanford.edu/~meeg/qick/fw/2022-02-18_qick216/). However, I meet runtime errors at the initial step of initializing QickSoc. The error information is as follows:

[image]

I have been a QICK user on ZCU111 and it was very successful. I switched to ZCU216 this week and meet this problem. Do you have any idea about what is going on here？

Thanks!

jramsey123, 2023-01-25:
Do you have the properly constructed and named TICSPRO files for those clock rates?

meeg, 2023-01-25:
Why that firmware image? I don't know that you are using the wrong image - maybe somebody pointed you to that one with a good reason - but I don't even remember what that image was for.

Just to make sure, you are running on the current version of the QICK library? I guess you have made some local changes, to make `QickSoc()` (no arguments) load this other file instead of the default . . . this is not necessary, you should revert those changes and use the `bitfile` parameter (https://qick-docs.readthedocs.io/en/latest/_autosummary/qick.qick.html#qick.qick.QickSoc). If you prefer to keep your changes, please post them here so I know what your code is doing.

The .hwh file that came with the firmware image must be in the same place with the same name. In your case that would mean `qick_216_meeg.hwh`. Correct?

You should not need to do anything with TICS if you are just trying to bring up a standard firmware.

zhizhenzhong, 2023-01-25:
Thanks @meeg and @jramsey123 for your prompt response. I will answer your questions:

@jramsey123 I git clone the most recent qick repo and did not change any TICSPRO files. So I assume it is the correct one on the repo?

@meeg Yes, I am running on the most recent QICK library. I made local changes but not major logic changes (only print the bitstream file name, etc for debug). You can have access to my code here and see my two recent commits https://github.com/zhizhenzhong/qick. I download the "qick_216.bit” and "qick_216.hwh“ from https://slac.stanford.edu/~meeg/qick/fw/2022-02-18_qick216/ and rename them to "qick_216_meeg" because there existing qick_216 files and I do not want to overlap with them. Note that even if I try the current qick_216.bit file downloaded with the repo it still ends up with the same error.

zhizhenzhong, 2023-01-25:
I think the wired part of the problem is: there seems to be a problem with the ZCU216 PYNQ image (downloaded from Sara's google drive) because it automatically thinks the board is ZCU111, not ZCU216. I have to modify it manually. I assume this may be the root cause? some part of the PYNQ image is still reading some ZCU111 clock configurations?

[image]

meeg, 2023-01-25:
OK, I want to be sure I understand - if you just clone the current repo and try to load QickSoc (which should use the default qick_216.bit) that also fails? Let's try to figure that out before trying to debug other firmware images (just because I upload something to my web space does not mean it is supported, or even works - most of these are for testing).

I agree there seems to be something wrong with your PYNQ image. Can you print the contents of `/home/xilinx/REVISION`?

zhizhenzhong, 2023-01-25:
Thanks. If I directly clone the repo and load QickSoc, it fails because it thinks this is an ZCU111 board, if I do not modify it to be ZCU216, shown as follows：

[image]

The contents of `/home/xilinx/REVISION` is

Release 2021_11_18 84cbf3e
Board 2021_11_18 0c01bc6 https://github.com/Xilinx/ZCU111-PYNQ/

Hmm, it says the PYNQ is ZCU111-PYNQ, maybe this is the reason?

meeg, 2023-01-25:
I think you did not load the correct SD card image.

zhizhenzhong, 2023-01-25:
Agreed. Is this (https://github.com/sarafs1926/ZCU216-PYNQ/issues/1) the correct location to download the ZCU216 image? or do you have other sources?

meeg, 2023-01-25:
That is the correct location. You should try again - it is not unheard of to copy the wrong image file onto an SD card, or even to plug the wrong SD card into a board.

zhizhenzhong, 2023-01-25:
I completely format my entire SD card, and burn the image again. Now it works. 

Let me share what I learnt and maybe it is useful. I figure out that the root cause is, I was using an old SD card which previous loads the PYNQ image for ZCU111. When I erase the card using Mac Disk Utility Tool, it only erase the PYNQ part but leave the file system of the ZCU111 linux in another partition of the card. So when I burn the ZCU216 PYNQ image to the card it creates a new partition, and somehow the PYNQ is still picking up the old ZCU111 linux file system. I need to completely wipe out every partition of the SD card. 

Thanks for helping me out. I will be exploring the ZCU216 board with my experimental setup.

closed 2023-01-25 by zhizhenzhong