openquantumhardware/qick #106: Multiplexing several readout in one ADC (ZCU111)

JavierSerranoGarcia, 2023-01-26:
Hi,
I want to multiplex one ADC  to readout several qubits. So  I have to send readout pulses at different frequencies through the same channel and analyze the received signal at the resonator frequencies. 

I have not found in Qick any method to do that. I wonder if I missed it reading the documentation or I have to implement everything. 

Best

meeg, 2023-01-26:
This can be done, and is supported by the software, but you need a special firmware. Try this, I hope the included demo notebook is enough to get you started: https://www.slac.stanford.edu/~meeg/qick/fw/2022-08-19_111_rfbv1_muxin/

Because this firmware was built for use with the QICK RF board and not the XM500 balun board that comes with the ZCU111, you should be careful to remove all the jumpers on the balun board before loading this firmware.

Alternatives (I am not particularly recommending these, these are just backup options that exist):
* use an RF splitter to send your single resonator signal to multiple ADCs
* use a different sort of special firmware which has a readout whose downconversion frequency can be changed dynamically inside the program, so you can time-multiplex your readout

JavierSerranoGarcia, 2023-01-27:
Thank you, Sho.

By removing all the jumpers you mean removing the black cap jumpers on both sides of the XM500 tagged J9 and J10, no?

As  we are planning to get ZCU216 as the next step, is there an overlay doing the same for this other board?

Thank you again.

meeg, 2023-01-27:
Yes, those jumpers.

Yes, we have ZCU216 firmware with this readout.

JavierSerranoGarcia, 2023-01-30:
Thank you so much.

closed 2023-01-30 by JavierSerranoGarcia