openquantumhardware/qick #112: Sending waveform at 10Gsps

JavierSerranoGarcia, 2023-02-22:
For some experiments, we would like to send a waveform at a maximum sampling rate 10Gsps (so a sampling period of around 0.1 ns). Is it possible in QICK?

meeg, 2023-02-22:
In terms of the sampling rate, you can get 9.5 Gsps on either the ZCU216 or the RFSoC4x2, though you will need to ask us for a different 216 firmware.

And you can send arbitrary waveforms using add_pulse and "arb" pulse style. But maybe you are asking for something more than that?

JavierSerranoGarcia, 2023-02-23:
What we would need is to add a pulse with the samples of the signal I want to send directly through the DAC, so without any modulation. Usually, you add pulses by giving the envelope and then modulating but in our case, we would like to send them directly (we want to send very short pulses with a special shape to the flux line of a qubit).

Hope this clarifies my question

meeg, 2023-02-23:
The standard way to disable the carrier modulation is to set `outsel='input'` in `set_pulse_registers()`: https://qick-docs.readthedocs.io/en/latest/_autosummary/qick.qick_asm.html#qick.qick_asm.QickProgram.set_pulse_registers

You could also set the carrier frequency to 0 to get a similar result, but this is not really the preferred solution.

JavierSerranoGarcia, 2023-02-24:
Thank you Sho, I forgot that parameter in set_pulse_register. But the other parameters related to the modulation (style, freq, phase) should contain any special value or just None?

meeg, 2023-02-24:
Style has the usual meaning, and what you want is "arb". The other parameters that you would normally specify for "arb" pulses must still be specified and not None, but the freq and phase will have no actual effect since they control the carrier. I suggest you experiment.

closed 2023-03-08 by JavierSerranoGarcia