openquantumhardware/qick #114: Special registers associated with a generator chanel

JavierSerranoGarcia, 2023-02-23:
The function for getting  special register number associated with a generator channel and register name only offer two options: "gain", "freq". 

Is there any other option like phase? 

```python
def sreg(self, gen_ch, name):
          
         """Gets tProc special register number associated with a generator channel and register name.

        Parameters
        ----------
        gen_ch : int
            generator channel (index in 'gens' list)
        name : str
            Name of special register ("gain", "freq")

        Returns
        -------
        int
            tProc special register number
       """

        tproc_ch = self.soccfg['gens'][gen_ch]['tproc_ch']
        return self._sreg_tproc(tproc_ch, name)
```

meeg, 2023-02-23:
Sorry, those names are just examples and not a complete list. The full list of possible names is here:

https://github.com/openquantumhardware/qick/blob/a469473a3dc502bfb05d6b42f0bbc7391b65dd84/qick_lib/qick/qick_asm.py#L1038

A given pulse will generally not use all of those registers. The registers are set by the set_reg() calls here:
https://github.com/openquantumhardware/qick/blob/a469473a3dc502bfb05d6b42f0bbc7391b65dd84/qick_lib/qick/qick_asm.py#L784-L856

JavierSerranoGarcia, 2023-02-24:
Got it, thank you so much.

closed 2023-02-24 by JavierSerranoGarcia