openquantumhardware/qick #126: Global phase control

rodolfocarobene, 2023-03-27:
Hello, I'm one of the developer of qibolab and I'm currently trying to execute a Ramsey experiment using the RFSoC4x2. 
Using the program present in the demo notebook I can do it properly, but with our driver doesn't seem to work.

In particular we are using this code:
````
# RAveragerProgram
class RamseyProgram(RAveragerProgram):
    def initialize(self):
        cfg=self.cfg
   
        self.q_rp=self.ch_page(cfg["qubit_ch"])     # get register page for qubit_ch
        self.r_wait = 3
        self.r_phase2 = 4
        self.r_phase=self.sreg(cfg["qubit_ch"], "phase")
        self.regwi(self.q_rp, self.r_wait, cfg["start"])
        self.regwi(self.q_rp, self.r_phase2, 0)
        
        self.declare_gen(ch=cfg["res_ch"], nqz=2) #Readout
        self.declare_gen(ch=cfg["qubit_ch"], nqz=2) #Qubit
        for ch in [0,1]: #configure the readout lengths and downconversion frequencies
            self.declare_readout(ch=ch, length=cfg["readout_length"],
                                 freq=cfg["f_res"], gen_ch=cfg["res_ch"])
        
        f_res=self.freq2reg(cfg["f_res"], gen_ch=cfg["res_ch"], ro_ch=0) # conver f_res to dac register value
        f_ge=self.freq2reg(cfg["f_ge"], gen_ch=cfg["qubit_ch"])

        self.set_pulse_registers(ch=cfg["qubit_ch"], style="const", freq=f_ge, phase=0,
                                 gain=cfg["pi2_gain"], 
                                 length=self.soccfg.us2cycles(cfg["len"], gen_ch=cfg["qubit_ch"]))
        self.set_pulse_registers(ch=cfg["res_ch"], style="const", freq=f_res, phase=cfg["res_phase"], gain=cfg["res_gain"], 
                                 length=cfg["readout_length"])        
        
        self.sync_all(self.us2cycles(0.2))
    
    def body(self):
        self.regwi(self.q_rp, self.r_phase, 0)
        
        self.pulse(ch=self.cfg["qubit_ch"])  #play probe pulse
        self.mathi(self.q_rp, self.r_phase, self.r_phase2,"+",0)
        self.sync_all()
        self.sync(self.q_rp,self.r_wait)

        self.pulse(ch=self.cfg["qubit_ch"])  #play probe pulse
        self.sync_all()

        #trigger measurement, play measurement pulse, wait for qubit to relax
        self.measure(pulse_ch=self.cfg["res_ch"], 
             adcs=[0,1],
             adc_trig_offset=self.cfg["adc_trig_offset"],
             wait=True,
             syncdelay=self.us2cycles(self.cfg["relax_delay"]))
        
    def update(self):
        self.mathi(self.q_rp, self.r_wait, self.r_wait, '+', self.cfg["step"]) # update the time between two π/2 pulses
        self.mathi(self.q_rp, self.r_phase2, self.r_phase2, '+', self.cfg["phase_step"]) # advance the phase of the LO for the second π/2 pulse

````
And executing:
````
t2p=RamseyProgram(soccfg, config)
x_pts, avgi, avgq= t2p.acquire(soc, load_pulses=True,progress=True, debug=False)

subplot(111, title="Ramsey Fringe Experiment", xlabel="Delay time [us]", ylabel="Qubit Population")
plot(soccfg.cycles2us(x_pts),np.abs(avgi[0][0]+avgq[0][0]*1j),'o-')
````
We obtain:
[image]

While using our driver we are running something similar to this code:
````
# AveragerProgram
class RamseyProgramSlow1(AveragerProgram):
    def initialize(self):
        cfg=self.cfg
        
        self.q_rp=self.ch_page(cfg["qubit_ch"])     # get register page for qubit_ch
        self.wait = cfg["wait"]
        
        self.declare_gen(ch=cfg["res_ch"], nqz=2) #Readout
        self.declare_gen(ch=cfg["qubit_ch"], nqz=2) #Qubit
        for ch in [0,1]: #configure the readout lengths and downconversion frequencies
            self.declare_readout(ch=ch, length=cfg["readout_length"],
                                 freq=cfg["f_res"], gen_ch=cfg["res_ch"])
        
        f_res=self.freq2reg(cfg["f_res"], gen_ch=cfg["res_ch"], ro_ch=0) # conver f_res to dac register value
        f_ge=self.freq2reg(cfg["f_ge"], gen_ch=cfg["qubit_ch"])
        self.f_ge = f_ge
        self.set_pulse_registers(ch=cfg["qubit_ch"], style="const", freq=f_ge, phase=0,
                                 gain=cfg["pi2_gain"],
                                 length=self.soccfg.us2cycles(cfg["len"], gen_ch=cfg["qubit_ch"]))
        self.set_pulse_registers(ch=cfg["res_ch"], style="const", freq=f_res, phase=cfg["res_phase"], gain=cfg["res_gain"], 
                                 length=cfg["readout_length"])        
        
        self.sync_all(self.us2cycles(0.2))
    
    def body(self):        
        self.pulse(ch=self.cfg["qubit_ch"], t=0)  #play probe pulse
        
        self.set_pulse_registers(ch=self.cfg["qubit_ch"], style="const", freq=self.f_ge, phase=self.cfg["phase"],
                                 gain=self.cfg["pi2_gain"], 
                                 length=self.soccfg.us2cycles(self.cfg["len"], gen_ch=self.cfg["qubit_ch"]))
        
        self.pulse(ch=self.cfg["qubit_ch"], t=self.soccfg.us2cycles(self.cfg["len"])+self.wait)  #

        #trigger measurement, play measurement pulse, wait for qubit to relax
        self.measure(pulse_ch=self.cfg["res_ch"], 
             adcs=[0],
             t=2*self.soccfg.us2cycles(self.cfg["len"])+self.wait,
             adc_trig_offset=2*self.soccfg.us2cycles(self.cfg["len"])+self.wait+self.cfg["adc_trig_offset"],
             wait=True,
             syncdelay=self.us2cycles(self.cfg["relax_delay"]))
        
````
And executing:
````
res = []
for i, w in enumerate(np.arange(0, config["step"]*config["expts"], config["step"])):
    config["wait"] = w
    config["phase"] = soccfg.deg2reg(np.degrees(1*i), 0)
    config["off"] = soccfg.deg2reg(np.degrees(1*i), 0)
    t2p=RamseyProgramSlow1(soccfg, config)
    avgi, avgq= t2p.acquire(soc, load_pulses=True,progress=False, debug=False)
    res.append(np.abs(avgi[0][0]+avgq[0][0]*1j))
    
plt.plot(soccfg.cycles2us(np.arange(0, config["step"]*config["expts"], config["step"])), res,'o-')
````
We obtain:
[image]

I suppose that the problem is that we use a series of AveragerProgram and not a RAveragerProgram so there is a global phase changing between the executions. Is there a way to compensate for this issue and to obtain the same result in both case?
I tried to look into the phrst argument for set_pule_register, but didn't manage to solve the problem.

Thanks a lot in advance for your help!

rodolfocarobene, 2023-03-27:
Thanks for the fast response. 
Yes, the two plot are done exactly with the code that I posted (I apologize if it's a bit of a mess) . 
Here it is the config dictionary:
````
config = {
 'res_ch': 0,
 'qubit_ch': 1,
    
 'readout_length': soccfg.us2cycles(600*0.001, gen_ch=0),
 'f_res': 7371.8,
 'res_phase': 0,
 'adc_trig_offset': 200,
 'res_gain': int(0.046*32000),
    
 'pi_gain': int(0.05*32000),
 'pi2_gain': int(0.05*32000/2),
 'f_ge': 5541.94,
 'len': 30*0.001,
    
 'relax_delay': 100_000*0.001,
    
 'start': 0,
 'step': soccfg.us2cycles(100*0.001),
    
 'phase_step': soccfg.deg2reg(np.degrees(1), 0),
    
 'expts': 100,
 'reps': 10,
 'rounds': 200}
````
I also thought that could be a problem on how I defined two different pulses from the same channel (using the t argument and calling set_pulse_registers both in initialize and in the body), but I think that I ruled that out.
The `reps` variable was also increased in some tries as well as all the length involved and the phase step.

meeg, 2023-03-29:
1. You have a bug in your AveragerProgram, where the first pi/2 pulse is defined by a set_pulse_registers() in the initialize() block, and this is overwritten by the set_pulse_registers() for the second pi/2 pulse. So after the first iteration of the loop, you are driving both of your pi/2 pulses with the same phase (no detuning). The RAveragerProgram does the detuning correctly. So you the RAveragerProgram measurement gives you a cosine*exponential where the cosine frequency comes from the detuning, and the AveragerProgram gives you something that is close to a pure exponential.
2. This is not really wrong, but why are you making your phase_step so big? You are setting it equal to a radian, and your oscillation is very rapid as a result.
3. Similarly I don't see anything wrong with how you are applying the delay in your AveragerProgram, but it seems like it would be easier to do it the same way as the RAveragerProgram does it?
Instead of:
```
        self.pulse(ch=self.cfg["qubit_ch"], t=0)  #play probe pulse
        
        self.set_pulse_registers(ch=self.cfg["qubit_ch"], style="const", freq=self.f_ge, phase=self.cfg["phase"],
                                 gain=self.cfg["pi2_gain"], 
                                 length=self.soccfg.us2cycles(self.cfg["len"], gen_ch=self.cfg["qubit_ch"]))
        
        self.pulse(ch=self.cfg["qubit_ch"], t=self.soccfg.us2cycles(self.cfg["len"])+self.wait)  #

        #trigger measurement, play measurement pulse, wait for qubit to relax
        self.measure(pulse_ch=self.cfg["res_ch"], 
             adcs=[0],
             t=2*self.soccfg.us2cycles(self.cfg["len"])+self.wait,
             adc_trig_offset=2*self.soccfg.us2cycles(self.cfg["len"])+self.wait+self.cfg["adc_trig_offset"],
             wait=True,
             syncdelay=self.us2cycles(self.cfg["relax_delay"]))
```
I think you could do:
```
        self.pulse(ch=self.cfg["qubit_ch"], t=0)  #play probe pulse
        self.sync_all(self.wait)
        
        self.set_pulse_registers(ch=self.cfg["qubit_ch"], style="const", freq=self.f_ge, phase=self.cfg["phase"],
                                 gain=self.cfg["pi2_gain"], 
                                 length=self.soccfg.us2cycles(self.cfg["len"], gen_ch=self.cfg["qubit_ch"]))
        
        self.pulse(ch=self.cfg["qubit_ch"], t=0)  #
        self.sync_all()

        #trigger measurement, play measurement pulse, wait for qubit to relax
        self.measure(pulse_ch=self.cfg["res_ch"], 
             adcs=[0],
             t=0,
             adc_trig_offset=self.cfg["adc_trig_offset"],
             wait=True,
             syncdelay=self.us2cycles(self.cfg["relax_delay"]))
```

rodolfocarobene, 2023-03-29:
Thank you very much @meeg! The problem was indeed the position of set_pulse_registers and now everything works as expected!

In will try to answer also your other questions:
  2. I set that phase just for demonstration purposes, in qibolab to do ramsey we use a more complex formula that computes the step phase from the number of oscillation to see
  3. The idea is to use a single averagerprogram (or raveragerprogram) for every possible experiment. 
  We pass to the qick program a qibolab PulseSequence object that contains the start time of each pulse so it's much easier to use the t argument (the other possibility is to compute the time as relative between pulses, but this become more complex for a multiqubit driver). 
  I'm not sure I completely explained myself, if you want you can have a look at [this PR](https://github.com/qiboteam/fpga-server/pull/2) with the 2 qick programs we are trying to use.

meeg, 2023-03-29:
Makes sense, thanks!

closed 2023-03-29 by meeg