openquantumhardware/qick #136: RuntimeError: frequency collision (multiplexing)

rodolfocarobene, 2023-04-12:
Hello, we are now trying multiplexed readout with the ZCU111 and sometimes, depending on the frequencies we are sending, we get this error (obviously frequencies change):
```
RuntimeError: frequency collision: tried to set PFB output 2 to 250.000002 MHz and output 1 to 150.000000 MHz, 
but both map to the PFB channel that is optimal for [96.000000, 288.000000]
(all freqs expressed in first Nyquist zone)
```
I tried to look into the documentation, but couldn't really understand it. Could you please explain me why it appears and eventually how to avoid it?

Thanks in advance!

meeg, 2023-04-12:
Sorry, the readouts and generators other than the "standard" ones (the ones used in the standard firmwares) are still in development and not very fully documented. But the error message is correct. The multiplexed readout uses a "polyphase filter bank" (PFB) to "channelize" the input signal into 8 frequency bands; depending on what frequencies you select, the appropriate PFB channels have the correct digital downconversion applied and get wired to the readout buffers. But in the current design it is not possible to apply two different downconversion frequencies to the same PFB channel, and that is what the error is telling you.

You must avoid using multiple frequencies that would map to the same frequency band. If you are using an analog mixer, you may be able to shift your IFs to map into different bands.

rodolfocarobene, 2023-04-13:
Ok thank you for the explanation. These frequencies bands are fixed and known? (Like 0-92, 92-288, ...) 
If they were we could probably take in account the bands correctly.

I have also a couple of other questions always regarding the multiplexed readout.
We were trying to do resonator spectroscopy with 3 different resonators  and we noticed that:

* if we set amplitude=1 and we use just one tone this has a different amplitude than 2 tones set at amplitude=1. It seems that there is a total amplitude that "gets splitted" in the active multiplexed channels (so two tones at amplitude=1 have half the amplitude of a single tone at amplitude one), is this correct? 
* a spectroscopy with 2/3 tones worked fine (a bit noisier than the single tone case but ok) with a number of shots ~100. When we increase the number of shots, without changing anything else (eventually increasing the final relax delay), we obtain nosier data and at ~1000 shots we do not see any resonance. This behaviour was a bit unexpected, could this besomehow related to the firmware?

meeg, 2023-04-14:
Yes, the band boundaries are (fs/32)*(2*i + 1), where fs is the ADC sampling frequency (3072 MHz in this case).

Yes, the multiplexed signal generator automatically scales the tone amplitude depending on the number of enabled tones, to avoid exceeding the DAC range.

I don't have an answer for the last question.

rodolfocarobene, 2023-04-24:
Sorry, I forgot of this open issue... 
Thank you very much for your explanations, now everything is working properly.

The problem with the shots was a weird one, but with easy solution, I'll try to explain it now for everyone that may encounter the same problem.
I was considering as multiple readouts the multiplexed one and setting the wrong value in the acquire function.
For example: if I had a multiplexed readout with 3 tones, I was setting `reads_per_rep` as 3, while actually the correct value was 1.