openquantumhardware/qick #139: ZCU216 DAC seems to not follow the external ref 10MHz clock.

tristanlorriaux, 2023-04-26:
Hi there !
My team and I are encoutering a big issue, since its causing trouble in all of our measurements. This issue is related with the phase of the DAC, which is not constant at each measurement. This is linked with the fact that the ZCU216 seems to not consider the external reference input (for instance, if I set a DAC to 10 MHz, its phase will be not the same as the reference). Futhermore, the DACs also don't follow the external output reference clock when it's activated. To sum up :

- DAC don't synchronize to the input reference clock when it's activated (`external_clk = True`) > J11 SMA Port 10 MHz TTL input
- DAC are not synchronized to the output reference clock (`clk_output = True`) J10 SMA Port 245.76 Output

BUT : DACs are synchronized hopefully beetween themselves. Have you an idea in order to solve this problem? 
Thanks a lot in advance ! 

Tristan from QCircuits, Lyon 🚀

meeg, 2023-04-27:
Are you correctly considering frequency quantization? The DACs have 32-bit frequency resolution, so they cannot generate exactly the frequencies you are expecting (neither 10 nor 245.76 is a round multiple of 6881.28/2**32). Your freq2reg commands generate 32-bit integers that correspond to the closest frequency to what you requested. You can use reg2freq (https://qick-docs.readthedocs.io/en/latest/_autosummary/qick.qick_asm.html#qick.qick_asm.QickConfig.reg2freq) to convert back to a frequency.

If that's not the explanation, then I think you need to give more specifics about your clocking and measurement setup, I do not really have a complete picture of what you are doing and seeing.

tristanlorriaux, 2023-04-28:
Hi, 
Thanks for your quick answer ! To give more details and context, here are two experimental setups we tested. 
**We want to use QICK as an IF source and a LO that we mix via an IQ mixer.** 

- The first setup works. We launch qick and we itilize the SoC with `output_clk=True`. The J10 port of the CLK106 provides a reference sinusoid at 245.76 MHz. If we set QICK and the LO to the same frequency (a multiple of  6881.28/2**32, checked with `reg2freq()`) and look at the Lissajous, it is fixed for several hours.

- The second setup, based on the input reference clock of QICK does not seem to work (and we think that it may be caused by a bug?). We generate a 10 MHz signal (TTL, or a 2Vpp sine wave, both don't work). This signal is given on port J11 of the ZCU and the SoC is initialized with `external_clk=True`. This external reference clock is also given as a reference to the LO. And in this case, the Lissajous pattern is not conclusive : the two signals do not seem to be in phase. But we have the proof that the LO is well locked on this external reference clock, so it seems that the problem comes from QICK. Do you have an idea of the source of our issue? 

Hoping to have given enough details and waiting for your answer !
Tristan 🚀

meeg, 2023-04-29:
We have done similar tests with good results. For your second test:
* You can see in the QICK paper (https://arxiv.org/pdf/2110.00557.pdf, appendix A) the result of an equivalent test, though with a ZCU111.
* We also did a more basic test of the ZCU216 `external_clk=True` setting where we locked the ZCU216 to a spectrum analyzer's 10 MHz reference output and verified that the DAC output as measured by the SA, which was previously off by O(ppm) and fluctuated with the temperature of the ZCU216, was now correct to maximum resolution.

But I can do the full test with the 216 next week.

meeg, 2023-05-01:
Works fine here. You can see that the Lissajous figure is quite stable, this ran for an hour.
[image]
I have an RF source, HMC-T2220. This makes a 10 MHz reference (about 1 V peak-to-peak, though the datasheet says 10 dBm typical), which goes to the ZCU216 J11, and an 84 MHz output, which goes to the scope. I have the QICK make 84 MHz CW (`style='const',mode='periodic'`), which also goes to the scope.

tristanlorriaux, 2023-05-02:
Hi ! 
Seems to finally works. The issue was a insufficient power going to the ZCU216's J11 (we were giving about 0dBm). Now the Lissajous is stable for hours. 

Thanks a lot 🚀
Tristan

closed 2023-05-02 by tristanlorriaux