openquantumhardware/qick #140: Multiplexed readout phase

rodolfocarobene, 2023-04-27:
Hello, I'm once again asking for your support :-)

We (@JavierSerranoGarcia and I) are trying to use the multiplexed readout and we noticed that, when we use it, the phase does not make much sense. I also think that this is reflecting in the amplitude, adding noise.

I'm attaching a [notebook](https://github.com/rodolfocarobene/qick/blob/main/qick_demos/Example%20problem%20multiplexed%20readout.ipynb) for clarity.

This is the phase for the standard bitstream (no multiplex)
[image]
and this is with the multiplex
[image]

The problem with the phase is present also when using a single tone (but with the multiplexed readout).

This is probably because of how I've written the qickprogram, but (at least in the notebook) I'm precisely using the same code that you gave us as example in #106 so I don't know what to change to correct it.

Thanks a lot in advance for your help!

meeg, 2023-04-29:
This behavior is expected and unavoidable due to how the multiplexed readout works - phase will not be repeatable after you change the frequency. See https://github.com/openquantumhardware/qick/issues/122. The amplitude should not be affected.

rodolfocarobene, 2023-05-01:
Ok, I see, thanks for your explanation.

Out of curiosity, is it possible that this issue will be fixed in the future or will it be a permanent limit of qick?

meeg, 2023-05-01:
It will be fixed at some point - neither limitation of the mux readout (#122 and #136) is fundamental, we know how they can be removed, we are just busy working on other parts of QICK.

I would emphasize that the mux generator and readout are not part of the standard QICK firmware. They are neither fully developed nor documented, there are other things like this that exist in the firmware source and are available on request, but are sort of provided at your risk. People have found them useful for reading out qubit states at frequencies that have already been determined using the standard gen+readout.

rodolfocarobene, 2023-05-02:
Okok, I understand.
Thank you for the explanations and all the work you are doing!

closed 2023-05-02 by rodolfocarobene