openquantumhardware/qick #141: Dynamic range control for ADC

mutasemodeh, 2023-04-30:
Hi,

I wonder if there is a feature of controlling the voltage sampling range of the QICK ADC for optimal dynamic range?

In the Zynq page, there is a the option of integrated digital step attenuator (DSA) and I wonder if this feature can be controlled by QICK overlay. 

https://docs.xilinx.com/r/en-US/pg269-rf-data-converter/Digital-Step-Attenuator-Gen-3/DFE

Thanks,
Mutasem

meeg, 2023-04-30:
Thanks, this is an interesting question - yes, this function is exposed by the Xilinx driver, so it's accessible to you even though the QICK code does not specifically hook into it. (We could do that? On the other hand I think it is somewhat unusual for fridge output powers to exceed 0 dBm, so zero attenuation is usually optimal anyway?)

This worked for me (I'm on a ZCU216, one of my readout channels uses tile 2 channel 2, I see that its readout amplitudes drop by sqrt(10) after I run this code):
```
the_adc = soc.rf.adc_tiles[2].blocks[2]
print(the_adc.DSA)
the_adc.DSA['Attenuation'] = 10.0
print(the_adc.DSA)
```
output:
```
{'DisableRTS': 0, 'Attenuation': 0.0}
{'DisableRTS': 0, 'Attenuation': 10.0}
```

meeg, 2025-03-05:
Cleaning up old issues; feel free to reopen if you have some follow-up.

closed 2025-03-05 by meeg