openquantumhardware/qick #144: Gain: DAC units mapping to dbm

EeshGupta, 2023-05-18:
Hi, I wanted to know whether there is a mapping from DAC units (0-30000) to units of power (dbm). Issue @82 addressed a similar question however the answer there seemed to be try-and-check diff values.

meeg, 2023-05-18:
No, there is no fixed mapping between DAC units and real units. For one thing, there are multiple components of the DAC response which are frequency-dependent.

I would characterize the answer to #82 diffferently - basically "there is a definite answer to this question, but it's easier to answer it in terms of end-to-end specifications than by reading the code" - but the answer to your question absolutely is "there is no universal answer to your question, it will depend on the signal you are driving, and therefore you need to test and measure your signal."

EeshGupta, 2023-05-18:
Got it, thanks!

closed 2023-05-18 by EeshGupta