openquantumhardware/qick #145: Determine Whether to Play Pulse or Not Based on Readout Results

TinaHsing, 2023-05-23:
I am a novice on the Qick platform and would like to know how to use tProc to decide whether to play a pulse or not based on the comparison of readout with a custom threshold. Your sample code demonstrates comparing a specific number with a threshold. How can I replace the number with the readout result?

TinaHsing, 2023-05-24:
Thanks for your fast response, I am sorry that I didn't check more detail

meeg, 2023-05-24:
You're right, that part of the tProcessor manual is out of date. `read i, p, lower/upper, $r` reads input `i`'s accumulated I/Q into page `p`, reg `r` - so `self.read(0, 0, "lower", 2)` reads input 0's I into page 0 register 2.

closed 2023-07-31 by TinaHsing