openquantumhardware/qick #149: Duration Sweepers

rodolfocarobene, 2023-06-01:
Hello everyone, is it possible to do hardware sweepers on durations?

I understand this is not trivial because something like that would also change the  waveforms, but a possible solution could be to load all the waveforms in `initialize()` and then call them "iteratively".
Is this something that is currently possible or the only option is python loops?

Also, different question always regarding sweepers, I saw in the [documentation](https://qick-docs.readthedocs.io/en/latest/_autosummary/qick.qick_asm.html#qick.qick_asm.QickRegisterManagerMixin.new_gen_reg) that is possible to call a register for `adc_freq`. Is this something that can be used for readout frequency sweepers?

As alwyas, thanks a lot for your help and your work

meeg, 2023-06-06:
To be clear, you are talking about sweeping the duration of arbitrary-waveform pulses? Correct, that's basically not possible, but certainly you could load a bunch of waveforms as you suggest.

Yes, it's possible to define a QickRegister for ADC frequency, but that is a software object which is only useful if you use a firmware with tProc-controlled readouts. Such a firmware is capable of frequency sweeps, but there are tradeoffs and this is not part of the standard firmware.

rodolfocarobene, 2023-06-09:
Ok, thanks as usual for the explanation!

closed 2023-06-09 by rodolfocarobene