openquantumhardware/qick #153: Clarification on intended behaviour of `prog.trigger()`

Cameron-Spence-Riverlane, 2023-06-21:
Hi,
I am trying to understand the API, and I have encountered something that seems to be either a bug or unintended behaviour. (ZCU216, latest distribution)

```
from qick import QickSoc
from qick import AveragerProgram

soc = QickSoc()

cfg = {"reps" : 1}
prog = AveragerProgram(soc, cfg)

pulse_time = prog.us2cycles(1)
prog.trigger(None, pins=[0], t=0, width=pulse_time)
```

Pin PMOD_0 is activated briefly (long enough to see on a multimeter) when `trigger` is called. From my understanding the expected behaviour is that `prog.trigger()` adds the `trigger` command to an `asm` program, which is only actually read by the device when calling an `acquire` or similar method. Here it appears that the trigger happens immediately; or, at some point in this script, the PMOD pin is activated unintentionally. Can you clarify if that is the case?

Thanks!

meeg, 2023-06-23:
Your understanding is correct, and it is strange that you're seeing a pulse even though your example never executes the program, or even uploads it to the FPGA - as you say, that would be done by `acquire`.

Maybe the pulse you're seeing is nothing to do with your program, it is happening when you initialize `QickSoc()`? That loads the firmware onto the FPGA, so you expect some transients as the FPGA boots. You should step through your example code to see which line is actually making the pulse. If you can look at it on a scope you could also see if it's the pulse you expect, or something longer (I wouldn't expect a typical multimeter to see a microsecond pulse).

Lufter, 2024-05-06:
Hi All, 
As a follow-up, I am wondering if external triggering through PMOD is only implemented for ADC. Is there a way to implement a similar functionality to the DACs with just pynq, without goining into VHDL? In our specific applications, we usually wish to trigger the DAC instead of the ADC to output a signal at a specific time specified. Thanks!

meeg, 2024-05-07:
I think you're misunderstanding what `trigger()` does - it is used to trigger a readout buffer and/or output a pulse on a PMOD pin: https://qick-docs.readthedocs.io/en/latest/_autosummary/qick.asm_v1.html#qick.asm_v1.QickProgram.trigger. It does not trigger a readout based on an incoming pulse to a PMOD pin.

You can set the QICK to wait for an incoming pulse on a PMOD pin before starting a program: https://qick-docs.readthedocs.io/en/latest/_autosummary/qick.qick_asm.html#qick.qick_asm.AcquireMixin.acquire, also #137  But that may not do everything you want.

jh1g, 2024-11-13:
> You can set the QICK to wait for an incoming pulse on a PMOD pin before starting a program: https://qick-docs.readthedocs.io/en/latest/_autosummary/qick.qick_asm.html#qick.qick_asm.AcquireMixin.acquire
Hi, this is also what I need, but is there an updated link to the page? The link above doesn't work for me.

meeg, 2024-11-13:
Sorry, we reconfigured our docs site, just remove the `en/`: https://qick-docs.readthedocs.io/latest/_autosummary/qick.qick_asm.html#qick.qick_asm.AcquireMixin.acquire

meeg, 2025-03-05:
Cleaning up old issues; feel free to reopen if you have some follow-up.

closed 2025-03-05 by meeg

NnguyenHTommy, 2025-08-20:
Hello! I am having a similar issue but more of a clarifying question. I would like to externally trigger the DAC output multiple times. Specifically I would like to: send external trigger to pmod+ -> DAC outputs a pulse sequence ->  send another external trigger -> DAC outputs a different pulse sequence. Is this possible or has anyone done this before? Looking at the acquire method from the link @meeg sent above, there seems to be a way to externally trigger multiple times between rounds. However, then the issue becomes changing the pulse sequence as I think acquire will initialize every time which I do not want to happen as this is slow.