openquantumhardware/qick #154: Fast reset and hardware classification

rodolfocarobene, 2023-06-23:
Hello, I was reading the notebook with the fast reset example and I had two questions:
* for what I understand, in a read instruction like `self.read(0,0,"lower",2)` we can read the I value with "lower" and the Q value with "upper" and and then apply some conditional logic on the measured value. However, for fast reset, one would need to classify 0-1 state applying a threshold *and* an angle, right? Is it possible to do this in the middle of a program?
* The numbers in the read instruction (for example 0-0-2) are the input, the page and the register related to the measurement, how can I find those? Is it something like "measurement number", "adc index", "variable page number"?

I'm not sure I explained myself properly, in any case thanks for the help!

meeg, 2023-06-23:
I think you missed that in the single-shot readout protion of that demo notebook (which, for future reference, is https://github.com/openquantumhardware/qick/blob/main/qick_demos/06_qubit_demos.ipynb), the readout tone phase (res_phase) has been calibrated to put the ground and excited states on either side of the +Q axis. So it's sufficient to apply a 0 threshold on the I value, and this is what the condj instruction is doing.

The syntax is `read(input_ch, page, upper/lower, reg)`.

rodolfocarobene, 2023-06-26:
Indeed I had overlooked that part, thank you!
Regarding #145, I had seen it but still I am not sure  to understand fully... The `page` and `reg` arguments are "arbitrary"?
And only the last measurement for that channel is available in this way, right?

meeg, 2023-06-26:
The tProcessor has 8 pages (0-7), each of which has 31 writable registers (1-31). Registers are like memory locations - you can write to any register, and then you can read (or make decisions based on) the value back from that register. You should be careful that the register you use is not being used elsewhere in your program (you can do `print(prog)`) to print your program in assembly format and see what registers are being used).

If this is not clear to you, you should read the manual - it is not 100% up to date, but is fine for understanding the operating principles of the tProc:

https://github.com/openquantumhardware/qick/blob/main/firmware/tProcessor_64_and_Signal_Generator_V4.pdf

rodolfocarobene, 2023-06-28:
Ok, thank you very much!
I think it's finally time to read the tProc manual :-)