openquantumhardware/qick #155: Possible mismatch in time registers name

rodolfocarobene, 2023-06-26:
Hello again, this time not with a question, but with a possible "bug" that we encountered in https://github.com/qiboteam/qibosoq/pull/45. 
I noticed that the unit conversion for sweepers on the time register where not working properly and I think this is caused by a mismatch in the names.
From our side, it was easy to patch, but in any case here it is.

For what I understand, these are the name of the standard registers of the channels.
https://github.com/openquantumhardware/qick/blob/d393103ff324265bd33d4ba3fcb25a4ef3534101/qick_lib/qick/qick_asm.py#L1360
So it is possible to do something like:
```python
register = self.get_gen_reg(ch, "t")
```
For a `QickSweep` however, when a new object is created, it checks the register type from here:

https://github.com/openquantumhardware/qick/blob/d393103ff324265bd33d4ba3fcb25a4ef3534101/qick_lib/qick/qick_asm.py#L16
and also here:
https://github.com/openquantumhardware/qick/blob/d393103ff324265bd33d4ba3fcb25a4ef3534101/qick_lib/qick/qick_asm.py#L2503

So, to do a "time sweeper" you actually need to change the register name, if you want to use the unit conversion  (that is pretty nice):
```python
register = self.get_gen_reg(ch, "t")
register.reg_name = "time"
new_sweep = QickSweep(self, register, start, stop, expts)
```

This does not happen with the other registers (at least the main ones) since the name are the same.
In any case it's not a big bug, but it would be nice if it was fixed in one of the new updates (if it is really a bug :-), not completely sure).

As always, thank you very much!

hatlabcz, 2023-06-27:
Hi @rodolfocarobene,

Thank you for the detailed report!

Actually, the names in `RegisterType` are not intended to fully match with the names in `pulse_registers`, as they are used for different purposes. [This line of code](https://github.com/openquantumhardware/qick/blob/d393103ff324265bd33d4ba3fcb25a4ef3534101/qick_lib/qick/qick_asm.py#L2651) makes it looks like they should match, but that's actually a coincidence when the name of the register also describes what type of value it represents.

The function `get_gen_reg` was designed for getting a QickRegister object that represents the register that will be passed to the "Generator" IP block on the FPGA, which controls the **shape** of the pulse. These registers include "freq", "phase", "addr", "gain" and "mode". The register `"t"` is not one of these registers that the "Generator" IP takes, instead, it is used by the tProc for scheduling the time of the pulse. Therefore, it makes sense for `"t"` to be in `pulse_registers`, but it is not designed to be gettable using `get_gen_reg`.

Currently the `get_gen_reg` function doesn't have specific confinement on what `"name"` can be passed to it, because different kind of generators sometimes use different names. When you do `get_gen_reg(ch, "t")`, you can indeed get the register in the tProc that controls the pulse triggering time. However, the sweep will not run properly. This is because, in the QICK python code, the triggering time of the pulse is set by the `QickProgram.pulse(ch, t='auto')` function. So when you call this function to play the pulse, like [here ](https://github.com/qiboteam/qibosoq/blob/b03e4c06675d77ff161fdbade9fdfa57770d12f2/src/qibosoq/programs/base.py#L188), the `"t"` register updated by the sweeper will be overwritten by the `t` you passed to `pulse()`.  Setting `pulse(ch, t=None)` can bypass the overwritten of the `"t"` register, but it will need some extra tweaks in the code to  make it actually work, also it makes it hard to keep track of the pulse triggering time at the software level...

A more 'QICK' way to run the triggering time sweep with the NDAveragerProgram would be:

In `initialize()` do

```python
# declare a new register that will be used for controlling the waiting time before the pulse is played
self.r_t = self.new_gen_reg(ch=ch, init_val=cfg["t_start"], reg_type="time", tproc_reg=True) 
# add sweep over the waiting time register
self.add_sweep(QickSweep(self, self.r_t, cfg["t_start"], cfg["t_stop"], cfg["t_expts"]))
```
The tag `tproc_reg=True` ensures that the unit conversion will be based on the tProc clock rate.

In `body()` do:

```python
self.sync(self.r_t.page, self.r_t.addr) # set the internal t_off to the value of the waiting time register (effectively, waiting for the period of time represented by self.r_t)
self.pulse(ch=ch, t="auto") # pulse will be played right after the waiting time
```
For more details of how the internal t_off works, you can check out the document [here](https://github.com/openquantumhardware/qick/blob/main/firmware/tProcessor_64_and_Signal_Generator_V4.pdf).

Hope this helps, and let me know if you have further questions!

-Chao

rodolfocarobene, 2023-06-28:
Thank you for the detailed answer!
I think I understand much better the underlying idea and I will definitely also read the tproc manual...

For now, I tried to work only with `self.pulse(ch, t=something)` avoiding the delay system since I find a bit more easy to keep track of the start of every, particularly when working with multiple channels/qubits at the same time.
But maybe I just needed a reason to convince myself :-)

>  Setting pulse(ch, t=None) can bypass the overwritten of the "t" register, but it will need some extra tweaks in the code to make it actually work

Can I ask you what kind of tweaks would be needed? 

Thanks again!

hatlabcz, 2023-06-28:
Actually I think the thing you wanted to do (if I understand it correctly) is a good example that shows the advantage of programming with the t_off  instead of defining pulse times that are always relative to absolute 0.

Say we want a pulse sequence like this:
  Expt1: pulse @ 1us
  Expt2: pulse @ 2us
  Expt3: pulse @ 3us
  Expt4: pulse @ 4us

Like we are sweeping the triggering time of the pulse relative to the starting time of each experiment. This can be done with the code I mentioned earlier. However, if you consider this whole experiment sequence with absolute time, then the 2nd pulse will happen at absolute time t=3 us, the 3rd pulse at t=6 us,  and the 4th pulse at t=10 us (assuming no idling time between experiments). So instead of doing a linear sweep over the waiting time register (1, 2, 3, 4), the absolute time register need to be swept over a "triangular number" series(1, 3, 6, 10), which is not impossible, but needs much more complicated code... 

Just to show the fixability of the code, here is an example, but you can see it's much more complicated...

In `initialize()`
```python
from qick.averager_program import QickSweep, AbsQickSweep, merge_sweeps

cfg = {
    "t_start": 1,  # [us]
    "t_stop": 4,  # [us]
    "t_expts": 4,
}

# calculate the step of the waiting time between pulses
t_step = (cfg["t_stop"]-cfg["t_start"])/(cfg["t_expts"]-1)

# register that represents the waiting time relative to last pulse
self.res_r_t_step = self.new_gen_reg(ch, init_val=t_step, reg_type="time", tproc_reg=True)
# linear sweep over the waiting time register 
t_step_sweep = QickSweep(self, self.res_r_t_step, t_step, t_step * cfg["t_expts"], cfg["t_expts"])

# get the absolute pulse time register of the generator channel
self.res_r_t = self.get_gen_reg(ch, "t")
# adjust register parameters to ensure the unit convert will be based on tProc clock
self.res_r_t.reg_type = "time"
self.res_r_t.gen_ch = None
self.res_r_t.ro_ch = None 

# create a sweep over the absolute pulse trigging time
time_sweep = AbsQickSweep(self, label="time_sweep")
time_sweep.expts=cfg["t_expts"]
# 'res_r_t_step' will be added to 'res_r_t' at each iteration, so we have a "triangular number" series for the absolute pulse time
time_sweep.update = lambda : self.res_r_t.set_to(self.res_r_t, "+", self.res_r_t_step)
time_sweep.reset = lambda : self.res_r_t.set_to(cfg["t_start"])

# make the sweep of the absolute pulse time and the waiting time register iterate together
self.add_sweep(merge_sweeps([time_sweep, t_step_sweep]))
```

In `body()`
```python
self.pulse(ch=ch, t=None)
```

Note that the above code also assumes that the whole sequence is only excuted once in the FPGA. If you also have repetition loops in the FPGA, then you need to consider that the "Expt1" in next repetition need to have an extra offset in the starting time (the finishing time of last experiment sequence)

Let me know if this makes sense : )

rodolfocarobene, 2023-06-29:
It totally makes sense!
I guess that I will have to change the timing to relative, but this should not be too difficult. :-)

Thank you very much for the help, it has been extremely useful!

closed 2023-07-01 by rodolfocarobene