openquantumhardware/qick #156: Adding more DACs and ADCs (full capacity of the ZCU216) ?

tristanlorriaux, 2023-06-27:
Hi there! 
The Quantum Circuits team and I are very happy with the ZCU's capabilities so far, and we're very grateful for all the work you're doing on the QICK project.
Just a question: are you planning to rebuild a firmware with 16 DACs and 16 ADCs (at full ZCU216 capacity) in the next few weeks? We're going to need it for future experiments. Otherwise, I understand that I can do it yourself with TCL files, but I'm not very familiar with Vivado, so if you have a firmware already built somewhere, that would be amazing !
Thanks a lot in advance :)

Tristan from QCircuits, Lyon 🚀

meeg, 2023-06-28:
Not in the next few weeks, I think that is rather overestimating the pace of progress!

Seven DACs is the maximum with the current architecture, since the tProcessor has eight outputs and one is reserved for trigger outputs (the next version will be more flexible, but this is still in development). There isn't a fundamental limit on ADCs.

tristanlorriaux, 2023-06-29:
Ok, thanks for the details ! 🚀
Tristan

closed 2023-06-29 by tristanlorriaux