openquantumhardware/qick #159: Unwanted triggering low voltage plateau

tristanlorriaux, 2023-06-29:
Hi there! 
We have a new question about QICK and our ZCU216. We can't figure out if it's a bug or a native feature. When we use `qick.qick_asm.QickProgram.trigger` (or its wrapped version in `measure()`), we measure a trigger pulse around 2V. But this value collapses after a few nanoseconds to a plateau of a few mV. The 'width' parameter seems to determine the length of this plateau and not the length of the plateau at 2V (below is an image of the trigger we observe). We need to keep the voltage of the first plateau. Is this normal operation (trigger not designed to last more than a few ns) or an error? 

[image]

Thanks in advance for your response ! 🚀
Tristan

meeg, 2023-06-30:
I think this is a problem with your scope setup. Your scope is configured for 50-ohm impedance, but the PMOD outputs are not meant to drive 50-ohm impedances (this is generally true of digital GPIO pins, such as on a Raspberry Pi), so you are overloading them. You can find information on the PMOD circuit here (it's the same for the ZCU111 and ZCU216):

https://docs.xilinx.com/r/en-US/ug1271-zcu111-eval-bd/User-PMOD-GPIO-Connectors

Use a high-impedance scope probe and you should see the correct 3.3 V pulse.

tristanlorriaux, 2023-07-03:
Hi,
Thank you for your quick reply!
Yes it seems to be an impedance matching problem. If I've understood correctly, it's customary for PMODs to have an output impedance of 200 Ohms. The problem is that we'd like to connect a system with 50 Ohm input impedance (pins are adapted here to BNC). I seem to see that there's an output after buffering on the circuit diagram, but I don't know if we can access it? If not, do you have any other idea of how to address a 50 Ohm system with these PMODs?

Thanks in advance for your response ! 🚀
Tristan

meeg, 2023-07-06:
I don't think 200 ohm is in any way a standard for PMOD - I guess you are getting that from https://digilent.com/reference/_media/reference/pmod/pmod-interface-specification-1_3_1.pdf but that is referring to a series resistor which would add to whatever is the bare output impedance of the driver circuit (and anyway, the circuit diagram shows that the ZCU216 does not have such a resistor).

If you look at the circuit diagram, you will see that the connector pin you're accessing is after the TXS0108E buffer chip (which shifts the output voltage levels). But you can see from the buffer's datasheet that it has an absolute max output current of 50 mA (3.3 V into 50 ohm is 66 mA), and it's really only designed to drive very small currents (sub-mA).

You need to add your own buffer, which is able to drive a 50-ohm load at whatever voltage your other electronics require. This is a common problem and there are many solutions depending on the situation, some resources:
* https://maker.pro/forums/threads/50-ohm-output-ideas.30327/
* https://www.sciencedirect.com/science/article/pii/S2405844021017771
* http://land-boards.com/blwiki/index.php?title=DRIVE50
* https://electronics.stackexchange.com/questions/122830/turn-ttl-output-of-logic-ic-into-output-capable-of-driving-50-ohm-devices
* https://electronics.stackexchange.com/questions/180679/difficulties-supplying-5v-ttls-to-50-ohm-terminated-single-photon-counter

closed 2023-07-12 by tristanlorriaux