openquantumhardware/qick #163: Using ZCY216 for controlling multiple flux qubits

JavierSerranoGarcia, 2023-07-10:
After using the RFSoC4x2 and the ZCU111 boards for control and readout of qubits now we are starting with the ZCU216. The first issues we found are

- The sampling frequency for the DACs is fs=6881.280 MHz. Can it be changed or is there a different version with a sampling frequency of around 9 GHz (like in RFSoC4x2)?
- Is there a multiplexed readout version of the firmware like in ZCU111?
- Is there a better solution for the flux DC signals than external differential amplifiers like in the ZCU111? (we are using the companion board that comes with the ZCU216).

I really appreciate any help you can provide.
Javi

meeg, 2023-07-13:
Maybe this will work for you: https://www.slac.stanford.edu/~meeg/qick/fw/2023-04-17_216_q3diamond_full688and958/

The DAC coupling works the same in the ZCU216 as in the ZCU111, you need external differential amplifiers in either case.

JavierSerranoGarcia, 2023-07-31:
Hi Sho,
Thank you for the new firmware. We are trying to use it but we do not know how to use the

	4 constant-IQ outputs:
	0:	DAC tile 1, ch 0, fs=6881.280 MHz
	1:	DAC tile 1, ch 1, fs=6881.280 MHz
	2:	DAC tile 1, ch 2, fs=6881.280 MHz
	3:	DAC tile 1, ch 3, fs=6881.280 MHz

Which channels should we use to send pulses?

meeg, 2023-08-01:
The constant-IQ outputs play constant tones. They are not what you want for pulses; you should use the regular generators. They are there, no?

JavierSerranoGarcia, 2023-08-01:
But in case we want to use them for something like a bias what should we do?

meeg, 2023-08-02:
OK: https://qick-docs.readthedocs.io/en/latest/_autosummary/qick.qick.html#qick.qick.QickSoc.set_iq

JavierSerranoGarcia, 2023-08-04:
Thank you so  much

closed 2023-08-04 by JavierSerranoGarcia