openquantumhardware/qick #166: [Errno 6] No such device or address

alec-yen, 2023-07-13:
Hi, we have 2 ZCU216's.  Both ZCU216's were installed in the same configuration.

Using the same micro SD card for both, we run the following command.
```
from qick import *
soc = QickSoc()
```

On the 1st ZCU216, it runs properly.
```
resetting clocks: 245.76 491.52

QICK configuration:

	Board: ZCU216

	Software version: 0.2.161
	Firmware timestamp: Thu Feb 17 16:46:52 2022

	Global clocks (MHz): tProcessor 430.080, RF reference 245.760

	7 signal generator channels:
	0:	axis_signal_gen_v4 - tProc output 1, envelope memory 65536 samples
		DAC tile 2, blk 0, 32-bit DDS, fabric=430.080 MHz, f_dds=6881.280 MHz
	1:	axis_signal_gen_v4 - tProc output 2, envelope memory 65536 samples
		DAC tile 2, blk 1, 32-bit DDS, fabric=430.080 MHz, f_dds=6881.280 MHz
	2:	axis_signal_gen_v4 - tProc output 3, envelope memory 65536 samples
		DAC tile 2, blk 2, 32-bit DDS, fabric=430.080 MHz, f_dds=6881.280 MHz
	3:	axis_signal_gen_v4 - tProc output 4, envelope memory 65536 samples
		DAC tile 2, blk 3, 32-bit DDS, fabric=430.080 MHz, f_dds=6881.280 MHz
	4:	axis_signal_gen_v4 - tProc output 5, envelope memory 65536 samples
		DAC tile 3, blk 0, 32-bit DDS, fabric=430.080 MHz, f_dds=6881.280 MHz
	5:	axis_signal_gen_v4 - tProc output 6, envelope memory 65536 samples
		DAC tile 3, blk 1, 32-bit DDS, fabric=430.080 MHz, f_dds=6881.280 MHz
	6:	axis_signal_gen_v4 - tProc output 7, envelope memory 65536 samples
		DAC tile 3, blk 2, 32-bit DDS, fabric=430.080 MHz, f_dds=6881.280 MHz

	2 readout channels:
	0:	axis_readout_v2 - controlled by PYNQ
		ADC tile 2, blk 0, 32-bit DDS, fabric=307.200 MHz, fs=2457.600 MHz
		maxlen 16384 (avg) 1024 (decimated)
		triggered by output 0, pin 14, feedback to tProc input 0
	1:	axis_readout_v2 - controlled by PYNQ
		ADC tile 2, blk 2, 32-bit DDS, fabric=307.200 MHz, fs=2457.600 MHz
		maxlen 16384 (avg) 1024 (decimated)
		triggered by output 0, pin 15, feedback to tProc input 1

	7 DACs:
		DAC tile 2, blk 0 is 0_230, on JHC3
		DAC tile 2, blk 1 is 1_230, on JHC4
		DAC tile 2, blk 2 is 2_230, on JHC3
		DAC tile 2, blk 3 is 3_230, on JHC4
		DAC tile 3, blk 0 is 0_231, on JHC3
		DAC tile 3, blk 1 is 1_231, on JHC4
		DAC tile 3, blk 2 is 2_231, on JHC3

	2 ADCs:
		ADC tile 2, blk 0 is 0_226, on JHC7
		ADC tile 2, blk 2 is 2_226, on JHC7

	4 digital output pins:
	0:	PMOD0_0_LS (output 0, pin 0)
	1:	PMOD0_1_LS (output 0, pin 1)
	2:	PMOD0_2_LS (output 0, pin 2)
	3:	PMOD0_3_LS (output 0, pin 3)

	tProc axis_tproc64x32_x8: program memory 8192 words, data memory 4096 words
		external start pin: None
```

On the 2nd ZCU216, however, it gives an `[Errno 6] No such device or address` error. The fact that we did not see this error on the 1st FPGA (using the same micro SD card) makes it seem like this is an error with the installation of the FPGA or that the FPGA is defective. Any suggestions for debugging this?

```
resetting clocks: 245.76 491.52
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-2-cf6eccdc61bb> in <module>
      1 # Load bitstream with custom overlay
----> 2 soc = QickSoc()
      3 # Since we're running locally on the QICK, we don't need a separate QickConfig object.
      4 # If running remotely, you could generate a QickConfig from the QickSoc:
      5 #     soccfg = QickConfig(soc.get_cfg())

/home/xilinx/jupyter_notebooks/qick/qick_lib/qick/qick.py in __init__(self, bitfile, force_init_clks, ignore_version, no_tproc, clk_output, external_clk, **kwargs)
    266             self.ip_dict['usp_rf_data_converter_0']['parameters'])
    267 
--> 268         self.config_clocks(force_init_clks)
    269 
    270         # RF data converter (for configuring ADCs and DACs, and setting NCOs)

/home/xilinx/jupyter_notebooks/qick/qick_lib/qick/qick.py in config_clocks(self, force_init_clks)
    428             self.download()
    429             if not self.clocks_locked():
--> 430                 self.set_all_clks()
    431                 self.download()
    432         if not self.clocks_locked():

/home/xilinx/jupyter_notebooks/qick/qick_lib/qick/qick.py in set_all_clks(self)
    565                 # default value is 0x012C22
    566                 xrfclk.xrfclk._Config['lmk04828'][lmk_freq][55] = 0x012C02
--> 567             xrfclk.set_ref_clks(lmk_freq=lmk_freq, lmx_freq=lmx_freq)
    568         elif self['board'] == 'RFSoC4x2':
    569             # master clock generator is LMK04828, always outputs 245.76

/usr/local/share/pynq-venv/lib/python3.8/site-packages/xrfclk/xrfclk.py in set_ref_clks(lmk_freq, lmx_freq)
    248 
    249     for lmk in lmk_devices:
--> 250         _set_LMK_clks(lmk_freq, lmk)
    251     for lmx in lmx_devices:
    252         _set_LMX_clks(lmx_freq, lmx)

/usr/local/share/pynq-venv/lib/python3.8/site-packages/xrfclk/xrfclk.py in _set_LMK_clks(lmk_freq, lmk)
    153         raise RuntimeError("Frequency {} MHz is not valid.".format(lmk_freq))
    154     else:
--> 155         _write_LMK_regs(_Config[lmk['compatible']][lmk_freq], lmk)
    156 
    157 def _get_spidev_path(dev):

/usr/local/share/pynq-venv/lib/python3.8/site-packages/xrfclk/xrfclk.py in _write_LMK_regs(reg_vals, lmk)
     71             data = struct.pack('>I', v)
     72             if lmk['num_bytes'] == 3:
---> 73                 f.write(data[1:])
     74             else:
     75                 f.write(data)

OSError: [Errno 6] No such device or address
```

meeg, 2023-07-13:
Hi - this is strange, it looks like the CPU is unable to communicate with the clock chips on the CLK104 daughterboard.

Just to make sure:
* you're using the ZCU216 image from https://github.com/sarafs1926/ZCU216-PYNQ/issues/1?
* you're not doing anything before trying to load the QickSoc? we did once see a similar error, where writing values to the fan controller somehow interfered with access to the clock chips - everything was fine if we rebooted and tried again without messing with the fan controller.
* all the jumpers, switches, etc. on the boards are configured identically?

Here are some things to try (probably in this order), to debug whether this might be a hardware problem:
* Check the connection of the CLK104 board to the ZCU216 - it seems possible that you'd get this problem if the connection was loose, or one of the pins on the high-density connector is bent/damaged.
* If the CLK104 board looks fine, you could swap your two CLK104 boards.
* Xilinx provides a special demo tool which exercises the clocks, DACs, and ADCs: https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/246153525/RF+DC+Evaluation+Tool+for+ZCU216+board+-+Quick+start - you can try running that on your two boards. If it fails on the bad board, you can use that as a starting point when dealing with Xilinx support, maybe you can get a warranty replacement.

alec-yen, 2023-07-16:
Hi, thanks so much for your help! Pinpointing the CLK104 helped a lot with debugging. You guessed correctly, it seems the CLK104 board was slightly loose. Making sure the connection is solid solved the problem - thanks again!

meeg, 2023-07-17:
Amazing, glad to hear it!

closed 2023-07-17 by meeg