openquantumhardware/qick #170: ADC reading reading is too small

TinaHsing, 2023-07-31:
Hi Dear developer
I used the zcu216 image download from [google drive](https://drive.google.com/file/d/1TID0e64Z_xC3YZiZgG4crdsa2XxCdLo_/view?usp=sharing)
Then I used the demo program in [00_Send_receive_pulse](https://github.com/openquantumhardware/qick/blob/main/qick_demos/00_Send_receive_pulse.ipynb) without any modification in **_LoopbackProgram_**.
my configure input is like below, I changed the pulse_gain to 30000
length to 100
readout length to 300

config={"res_ch":6, # --Fixed
        "ro_chs":[0], # --Fixed
        "reps":1, # --Fixed
        "relax_delay":1.0, # --us
        "res_phase":0, # --degrees
        "pulse_style": "const", # --Fixed
        
        "length":100, # [Clock ticks]
        # Try varying length from 10-100 clock ticks
        
        "readout_length":300, # [Clock ticks]
        # Try varying readout_length from 50-1000 clock ticks

        "pulse_gain":30000, # [DAC units]
        # Try varying pulse_gain from 500 to 30000 DAC units

        "pulse_freq": 100, # [MHz]
        # In this program the signal is up and downconverted digitally so you won't see any frequency
        # components in the I/Q traces below. But since the signal gain depends on frequency, 
        # if you lower pulse_freq you will see an increased gain.

        "adc_trig_offset": 0, # [Clock ticks]
        # Try varying adc_trig_offset from 100 to 220 clock ticks

        "soft_avgs":100
        # Try varying soft_avgs from 1 to 200 averages

       }

###################
# Try it yourself !
###################

prog = LoopbackProgram(soccfg, config)
iq_list = prog.acquire_decimated(soc, load_pulses=True, progress=True, debug=False)

I found :
A. the ADC reading is very small. and the rising edge is at t = 60(+100)
[image]
I double check the ouput waveform in scope, ch6 had around 800mVpp output. and I used xm655 card acompany with zcu216.
the balum selection is 10M~1G.
when I changed the pulse_freq  to 800 MHz, the reading is bigger but still small.
[image]

B. when i change the adc_trig_offset from 100 to 180 
[image]
we still can see the rising edge of output wavefrom at t = 0(+180)

Did I do anything wrong so that my adc reading is so small? 
the adc_trig_delay seems didn't have a good match to real waveform?

meeg, 2023-08-01:
Let me repeat to maike sure I understand correctly:
A. You're seeing very little signal in loopback, you have checked that your baluns are appropriate for the frequency and the output is visible on a scope.
B. You changed adc_trig_offset by 80 units, but the pulse did not shift by 80 units - more like 60.

and answers:
A. I agree you should be seeing a bigger pulse. You should double-check that you're connected to the correct ADC and that you have no loose connections there. You could also connect a signal generator to the ADC instead of the loopback cable, and play a constant sine wave at your readout frequency: since the generator will not be synchronized to the QICK you will not see a coherent constant value, but you should see something.
B. This is OK. `adc_trig_offset` is in units of the tProcessor clock, which is 430.08 MHz; the decimated data is sampled at the readout fabric clock, which is 307.2 MHz. So it makes sense that changing `adc_trig_offset` by 80 would shift the decimated data by 80*(307.2/430.08) = 57.1 ticks. (`soccfg.us2cycles(soccfg.cycles2us(80), ro_ch=0)` is a way to do this conversion in code.) If you look at the "Clocks and durations" section in the 00 demo you will see some more explanation of this.

meeg, 2025-03-05:
Cleaning up old issues; feel free to reopen if you have some follow-up.

closed 2025-03-05 by meeg