openquantumhardware/qick #182: ZCU216 image has no static MAC address

dangernoodle-dev, 2023-09-07:
Every time the ZCU216 is rebooted the MAC address changes. My fix is to put at static ip address in the /etc/network/interfaces.d/eth0 file via 

```
auto eth0
iface eth0 inet dhcp
    hwaddress ether 00:11:22:33:44:55

...
```
But this is a bit clumsy and maybe there's a way to fix the auto MAC detection?

meeg, 2023-09-07:
Yes, this is a known bug (see https://support.xilinx.com/s/question/0D52E00006x0pCJSAY/zcu216-mac-address-at-different-eeprom-offset-0x2f-instead-of-0x20?language=en_US), and what you're doing is the correct workaround.

This problem happens on our PYNQ 2.7 image for ZCU216 and the Xilinx PYNQ 3.0.1 image for ZCU111. We expect that the next round of SD images (our 3.0.1 image for ZCU216, and the Xilinx 3.1 image for ZCU111) will not have this bug.

It is possible to fix the autodetection: download the appropriate "bootfiles" tarball from https://www.slac.stanford.edu/~meeg/qick/bootfiles/ and copy the contents into `boot`:
```
tar -xzf bootfiles_v2pt7_ZCU216.tar.gz
sudo cp bootfiles_v2pt7_ZCU216/* /boot
```

cplonski20, 2024-08-15:
Hi Meeg,

Will you be releasing a PYNQ 3.0.1 image for the ZCU216 sometime soon?

Best,
Charlie

meeg, 2025-03-05:
Cleaning up old issues; feel free to reopen if you have some follow-up.

closed 2025-03-05 by meeg