openquantumhardware/qick #185: ADC readouts corrupted with non white noise

JavierSerranoGarcia, 2023-09-11:
Hi Sho

We are finding a weird behavior in the readouts and, maybe, you could shed some light on it. So, in a resonator spectroscopy experiment, we get something like this

[image]

It looks like there is a lot of noise disturbing the readout. But if you have a look in detail

[image]

you see it is not white noise but a very fast sinusoidal with a low sample rate (I think so).

Maybe you have found this issue previously or have some idea of what is going on.

Thanks

PD.: We are using the ZCU216 with multiplexed readouts

circuitqed, 2023-09-11:
One thing I've observed with all ADCs is that if you do a lot of averaging
(typically >50k or so) then you tend to start to see the clock in your
signal.  If you are sweeping at a uniform rate you could be seeing this or
an aliased version of it.

meeg, 2023-09-11:
I don't have time to dig into this, but I can guess:
* How big is your signal? Is it perhaps O(1) ADC unit? If so, you should be amplifying your signal more before going to the ADC - both because of the bug I describe next (which could explain your problem), but also because of quantization noise (which will just look like white noise). It is standard practice when driving any digitizer that you want your signal to be significantly larger than the ADC resolution, because of quantization noise.
* Plot the IQ values. As you change frequency, the complex IQ should rotate around the origin. It's possible that you will find that it instead rotates around (-0.5, -0.5) ADC units or something like that, due to a bug that we have seen in other firmware blocks. A constant offset like that will look like the sinusoidal oscillation you see here. If so, you should add (0.5, 0.5) before taking the magnitude.

JavierSerranoGarcia, 2023-09-12:
Thank you, Dave and Sho

I will follow your suggestion