openquantumhardware/qick #190: timming-violation for both zcu111 and zcu216 in 87895db

CCvitaminHsieh, 2023-09-19:
Hi,

We downloaded your repo (87895db) and run implementation for both zcu111 (proj_111.tcl) and zcu216 (proj_216.tcl) and we encountered timing violation in both cas: 

[image]

[image]

The vivado version we are using is 2022.1.

meeg, 2023-09-23:
Sorry for the delay, but this took some time:

When I generated the bitstreams from these projects originally, they passed timing. In response to your question I recreated and reran the projects, and now I see timing violations, but more severe than what you see. My suspicion is that this is related to the OS upgrade I ran on my PC in the interim (from Debian 11 to Debian 12); Vivado timing closure is supposed to be deterministic with respect to repeated runs, but not with respect to OS versions (https://support.xilinx.com/s/article/61599).

However I was able to get both projects to pass timing by changing the implementation strategy to Performance_Explore. So hopefully this will also work for you.

CCvitaminHsieh, 2023-09-25:
Thanks for your reply.

However, even if I changed the implementation strategy to "Performance_Explore (Vivado Implement 2022)", I still got "Failed Timing" as shown in the image below.
[image]
[image]

[image]

I encountered a 'timing violation' while running on Windows 10 22H2. Perhaps I could try changing the operating system to Debian or another Linux distribution to see if it resolves this issue.

meeg, 2023-09-26:
For what it's worth - if your worst negative slack is 12-13 ps like in your original screenshots, we think that's fine. Our firmware designer says anything under 50-60 ps is not something to worry about.

CCvitaminHsieh, 2023-09-28:
Ok. Thanks!

CCvitaminHsieh, 2023-09-28:
Ok. Thanks!

closed 2023-09-28 by CCvitaminHsieh