openquantumhardware/qick #197: A question regarding reset value of fifo_empty_r in axi_mst_write.sv

Ri-chard-Wu, 2023-10-16:
In the axi_mst_write.sv file, should the fifo_empty_r signal be reset to 1? since otherwise the first data that is written to DDR (when state <= DATA_ST) will always be 0, which is the reset value of fifo_dout_r?

[image]

[image]

leoeltipo, 2023-10-16:
Hello @Ri-chard-Wu 

Thanks a lot for checking so deep in the code!! You are totally right. This is just 1 latency register for the empty flag of a FIFO and I just ctrl+c and ctrl+v the line from the previous one and changed the name :)

We'll fix it and set it to the right value.

Thanks for that!!

Ri-chard-Wu, 2023-10-16:
Glad I could help!

closed 2023-10-16 by Ri-chard-Wu