openquantumhardware/qick #212: Unable to build my own firmwire

lisafan122, 2023-12-30:
set_property -name "dsa.emu_dir" -value "emu" -objects $obj

ERROR: [Common 17-142] Invalid property name 'dsa.emu_dir'.  Please use the 'list_property' command to find properties supported by the target 'project' object.

I run this proj_111.tcl but it gives me errors. Could you please check?

My ultimate goal is to generate the bitstream

meeg, 2023-12-30:
It works fine for me. Can you give the full console log, and confirm that you are running Vivado 2022.1?

lisafan122, 2023-12-30:
I am running on Vivado 2022.2. Oh! Do I need to change the version? thanks

meeg, 2023-12-30:
Yes, that project script was generated on 2022.1 and will not work on any other version. If you look for the block design script that's called by the project script, you will see the Vivado version there: https://github.com/openquantumhardware/qick/blob/ed694c55291d2c6bc0f7827ee575f6db52c6bdeb/firmware/proj_111.tcl#L77

closed 2023-12-31 by lisafan122