openquantumhardware/qick #213: change ADC readout port with bitstream

lisafan122, 2023-12-31:
Hi,

I have a by default ZCU111 with ADC224 ports to read. But I want to use 226 or 227. I have bitstream files built from the previous commit in this repo. But when I check the ADC port using QickSoc(), it still gives me the default setup. 

I perhaps forgot to change something aside from using that commit repo. Can you please suggest some?

Best,
Jaz

meeg, 2023-12-31:
I don't think I understand. What commit are you talking about? The ZCU111 projects in this repo have always placed two readouts on ADC224.

lucyfan7266, 2023-12-31:
Hi @meeg ,thank you for reply. I have used Vivado on proj_111.tcl (with bd_2022_2.tcl inside), generate a new bitstream .bit, and .hwh file and I want to load this bitstream into the ZCU111 board, so that my ADC ports are switched to 226. What is the right way to load the new bitstream into the ZCU111 board? I am using qick and QickSoc() will output the port descriptions. I hope to change the port according to my new bitstream. thanks

meeg, 2024-01-02:
Duplicate of #216

closed 2024-03-13 by meeg