openquantumhardware/qick #215: incomplete files in building new firmwire

lucyfan7266, 2023-12-31:
ERROR: [Vivado 12-172] File or Directory '/Directory/hdl/vect2bits_16.v' does not exist

Is there a location that I can find the file? Thanks

meeg, 2023-12-31:
It's this file: https://github.com/openquantumhardware/qick/blob/main/firmware/hdl/vect2bits_16.v

but you should not be getting this error if you followed the instructions in the README. Maybe your working directory is not the directory with the project script. If that doesn't help - please give me the full tcl console log from running the project script. Probably there is an earlier error or warning that tells you what the real problem is.

closed 2024-01-01 by lucyfan7266

meeg, 2024-01-02:
Duplicate of #212