openquantumhardware/qick #221: use a machine learning training algorithm from a different FPGA

lucyfan7266, 2024-01-13:
Hi team,

I try to run MNIST inference an training with ZCU111 board. I found code from another [ZipML](https://github.com/fpgasystems/ZipML-PYNQ/) repo, my current ZCU111 board has installed qick, could you inspire me how to write a code for training and inference in MNIST?
Thank you.
[image]

meeg, 2024-01-14:
Sorry, but ZipML seems to be software and firmware for the PYNQ-Z1 board. The only thing in common with QICK is the PYNQ OS. You would need to port the ZipML firmware to the ZCU111 and probably also make adjustments to the software; if you wanted to use ZipML functionality with QICK you would need to combine the two firmware designs.

closed 2024-01-14 by meeg

lucyfan7266, 2024-01-15:
Thank you. This is a reasonable solution! But I wanted to know, if I have implemented MNIST inference with ZCU111, can I use the training method in ZipML to update the MNIST training? Sorry I need some guidances.

meeg, 2024-01-15:
I don't know, and I don't see the connection to QICK. I can't help.