openquantumhardware/qick #236: Read jump point and error of direct downconversion

JayChien04717, 2024-02-21:
Hi meeg and  QICK team,
I face two problem on readout signal.

First, when I using DAC direct ouput high frequency and using a external LO to do down conversion.  After using ADC to readout. All instrument are 10MHz reference locked.
[image]
It seems phase are not locking cause when I increase average the signal amp decrease. Also repeat the `acquire_desimate`, the IQ signal are not fix.

Second is jump point,.
[image]
We change another connection to doing up conver and down convert (This way can keep phase lock). When we doing broad band spectrum scanning. There are some unreasonable data point. We try to change LO frequency, LO power and ADC readout frequency. But it still happened
[image]
[image]

If need more data to check please tell me. 
Very thanks for your help.

Jay

meeg, 2024-02-27:
Hi, sorry for the delay, I was on vacation last week.

First part: in principle this scheme, with analog mixing on the ADC side but not the DAC side, should work. However:
* it's really important that the frequencies sum exactly, right? In a symmetric setup (mixer on both sides, or neither side), you need the DAC and ADC frequencies to match exactly, otherwise you will see a rotating phase - as explained in the loopback demo. In your setup you need the DAC frequency to match exactly the sum of the LO and ADC, otherwise you will see a rotating phase - same problem. I suspect that's what you have here, even a sub-Hz frequency mismatch will ruin your averaging. With external RF generators it is not always easy to know what step size the frequency you request is getting rounded to; do you know the precise frequencies you are using on your DAC, ADC, and LO?
* Is there a reason you're doing this? The readout works OK even at fairly high undersampling. A mixerless setup may perform similarly to your setup even though the ADC will be operating in the 5th Nyquist zone.

Second part: it's very strange that the signal seems to periodically go to -2, even though in the good measurements the amplitude is fluctuating (in other words, the bad points always look the same regardless of the frequency). This doesn't look like any problem I recognize. I would suspect some bug in your Python code. Can you inspect the values you are getting at the bad points in the sweep - both the averaged values you are plotting, and the `prog.di_buf`/`prog.dq_buf`?

JayChien04717, 2024-03-01:
Thanks for your reply
Ya, I found that the frequency has slightly difference with my setting. So I try to conver dac frequency freq to reg and convert back to frequency and let ADC to recieve this value. But it still not very precisely when I using oscilloscope to check. I think it's the reason why the pase isn't lock.
Because I want to reduce the rf component using. I didn't use 5 nyquest zone to readout cause our signal is too noise and I guess 5 nyquest zone can't get the enough amplitude. But I will try later.

I will check `prog.di_buf`/`prog.dq_buf` later. Thanks for your seggestion. And I will close thie issue if there is no other problem.

closed 2024-03-01 by JayChien04717