openquantumhardware/qick #259: B_PHASE class attribute missing on AxisSgMux4Vx classes

dmtvanzanten, 2024-04-19:
The series of [AxisSgMux4Vx](https://github.com/openquantumhardware/qick/blob/ab60fe2b86c20e696c8b6ca100800136d1c9777c/qick_lib/qick/drivers/generator.py#L415) classes is missing the B_PHASE class attribute which makes calls to ```.deg2reg()``` failing with 
```python
File .....\qick_lib\qick\qick_asm.py:504, in QickConfig.deg2reg(self, deg, gen_ch, ro_ch)
    502 if ch_cfg is None:
    503     raise RuntimeError("must specify either gen_ch or ro_ch!")
--> 504 b_phase = ch_cfg['b_phase']
    505 return to_int(deg, 2**b_phase/360, parname='phase') % 2**b_phase

KeyError: 'b_phase'```

NOTE: the [AxisSgMux8V1](https://github.com/openquantumhardware/qick/blob/ab60fe2b86c20e696c8b6ca100800136d1c9777c/qick_lib/qick/drivers/generator.py#L447) class does have the B_PHASE class attribute.

meeg, 2024-04-19:
Are you calling deg2reg directly? There's a check in declare_gen() which should block you from defining phases for those generators and I think that should be the only place you'd try to do it for a muxed generator, so I want to understand how you got here.

dmtvanzanten, 2024-04-19:
I am calling deg2reg directly indeed. I think I may have ended up here due to being used to do this for other generators when I'm using there phase in register (Ramsey stuff etc...)

closed 2024-04-19 by dmtvanzanten