openquantumhardware/qick #269: Do the decimated IQ samples from the ADC have the wrong phase relationship?

LukeTrainor, 2024-07-10:
Basic summary: When acquiring samples with the ADC, in order to get the correct frequencies I have to take the Fourier transform of `I-1j*Q` instead of `I+1j*Q`, which I would expect.

The detail:

Hello, I am using qick on a 4x2 board. I usually use a different firmware, but for this example I'll use the default. This is the setup:
```
QICK configuration:

	Board: RFSoC4x2

	Software version: 0.2.260
	Firmware timestamp: Wed Sep  6 18:49:29 2023

	Global clocks (MHz): tProcessor 409.600, RF reference 491.520

	2 signal generator channels:
	0:	axis_signal_gen_v6 - envelope memory 65536 samples (6.667 us)
		fs=9830.400 MHz, fabric=614.400 MHz, 32-bit DDS, range=9830.400 MHz
		DAC tile 0, blk 0 is DAC_B
	1:	axis_signal_gen_v6 - envelope memory 65536 samples (6.667 us)
		fs=9830.400 MHz, fabric=614.400 MHz, 32-bit DDS, range=9830.400 MHz
		DAC tile 2, blk 0 is DAC_A

	2 readout channels:
	0:	axis_readout_v2 - configured by PYNQ
		fs=4423.680 MHz, decimated=552.960 MHz, 32-bit DDS, range=4423.680 MHz
		maxlen 16384 accumulated, 1024 decimated (1.852 us)
		triggered by output 7, pin 14, feedback to tProc input 0
		ADC tile 0, blk 0 is ADC_D
	1:	axis_readout_v2 - configured by PYNQ
		fs=4423.680 MHz, decimated=552.960 MHz, 32-bit DDS, range=4423.680 MHz
		maxlen 16384 accumulated, 1024 decimated (1.852 us)
		triggered by output 7, pin 15, feedback to tProc input 1
		ADC tile 0, blk 1 is ADC_C

	8 digital output pins:
	0:	PMOD0_0
	1:	PMOD0_1
	2:	PMOD0_2
	3:	PMOD0_3
	4:	PMOD0_4
	5:	PMOD0_5
	6:	PMOD0_6
	7:	PMOD0_7

	tProc axis_tproc64x32_x8: program memory 1024 words, data memory 1024 words
		external start pin: PMOD1_0

	DDR4 memory buffer: 1073741824 samples (1.942 sec), 256 samples/transfer
		wired to readouts [0, 1]

	MR buffer: 8192 samples (1.852 us), wired to readouts [0, 1]
```

To demonstrate, here's a minimal example based off of the tutorial `00_Send_receive_pulse.ipynb`. The only difference is the readout frequency is set separately. So we send a pulse at 500 MHz to the ADC reading out at 400 MHz. The generator channel is directly connected to the readout channel.

```
# Import the QICK drivers and auxiliary libraries
from qick import *
import numpy as np
import matplotlib.pyplot as plt
%matplotlib inline

# Load bitstream with custom overlay
soc = QickSoc()
soccfg = soc

GEN_CH = 0
RO_CH = 0

class LoopbackProgram(AveragerProgram):
    def initialize(self):
        cfg=self.cfg   
        res_ch = cfg["res_ch"]
        self.declare_gen(ch=cfg["res_ch"], nqz=1)
        
        # Set the readout frequency different to the pulse frequency
        for ch in cfg["ro_chs"]:
            self.declare_readout(ch=ch, length=self.cfg["readout_length"],
                                 freq=self.cfg["readout_freq"], gen_ch=cfg["res_ch"])
        
        freq = self.freq2reg(cfg["pulse_freq"],gen_ch=res_ch, ro_ch=cfg["ro_chs"][0])
        phase = self.deg2reg(cfg["res_phase"], gen_ch=res_ch)
        gain = cfg["pulse_gain"]
        self.default_pulse_registers(ch=res_ch, freq=freq, phase=phase, gain=gain)
        style=self.cfg["pulse_style"]
        self.set_pulse_registers(ch=res_ch, style=style, length=cfg["length"])
        self.synci(200)  # give processor some time to configure pulses
    
    def body(self):
        self.measure(pulse_ch=self.cfg["res_ch"], 
                     adcs=self.ro_chs,
                     pins=[0], 
                     adc_trig_offset=self.cfg["adc_trig_offset"],
                     wait=True,
                     syncdelay=self.us2cycles(self.cfg["relax_delay"]))

config={"res_ch":GEN_CH, # --Fixed
        "ro_chs":[RO_CH], # --Fixed
        "reps":1, # --Fixed
        "relax_delay":1.0, # --us
        "res_phase":0, # --degrees
        "pulse_style": "const", # --Fixed
        "length":200, # [Clock ticks]        
        "readout_length":300, # [Clock ticks]
        "pulse_gain":3000, # [DAC units]
        "pulse_freq": 500, # [MHz]
        "adc_trig_offset": 100, # [Clock ticks]
        "readout_freq": 400, # [MHz]
        "soft_avgs":1
       }

prog =LoopbackProgram(soccfg, config)
iq_list = prog.acquire_decimated(soc, progress=True)

iq = iq_list[0]
fADC = soccfg['readouts'][RO_CH]['f_fabric']
ft = np.fft.fftshift(np.fft.fft(iq[0]+1j*iq[1]))
fDecimated = np.linspace(-fADC/2,fADC/2,len(ft),endpoint=False)
plt.plot(fDecimated,np.abs(ft))
plt.ylabel("a.u.")
plt.xlabel("f (MHz)");
```

The generated graph is below, and it shows a peak at -100 MHz, whereas I believe it should be at 100 MHz.
[image]

Now the reason I think this might be a bug, instead of a sign convention is that if I send I and Q samples to the DAC, they have the sign convention I would expect. Here I set the pulse and readout frequency to both be at 500 MHz. The waveform from the I and Q samples is a complex exponential at 100 MHz. If I take the Fourier transform of the input IQ samples, there is a peak at 100 MHz. The measured samples, however, show a peak at -100 MHz.

```
pulse_length = 2**10
pulse_length_us = soccfg.cycles2us(1, gen_ch=0)*pulse_length/16 # 16 IQ samples per generator fabric clock tick
pulse_f = 100 # [MHz]
pulse_t = soccfg.cycles2us(1., gen_ch=0)*np.arange(pulse_length)/16
pulse_I = 32766*np.cos(2*np.pi*pulse_f*pulse_t)
pulse_Q = 32766*np.sin(2*np.pi*pulse_f*pulse_t)

class IQOffsetProgram(AveragerProgram):
    def initialize(self):
        cfg=self.cfg   
        res_ch = cfg["res_ch"]
        
        phase = self.deg2reg(self.cfg["res_phase"], gen_ch=self.cfg["res_ch"])
        freq = self.freq2reg(self.cfg["pulse_freq"], gen_ch=self.cfg["res_ch"], ro_ch=self.cfg["ro_chs"][0])

        # set the nyquist zone
        self.declare_gen(ch=cfg["res_ch"], nqz=1)
        
        # configure the readout lengths and downconversion frequencies (ensuring it is an available DAC frequency)
        for ch in cfg["ro_chs"]:
            self.declare_readout(ch=ch, length=self.cfg["readout_length"],
                                 freq=self.cfg["readout_freq"], gen_ch=cfg["res_ch"])

        # convert frequency to DAC frequency (ensuring it is an available ADC frequency)
        freq = self.freq2reg(cfg["pulse_freq"],gen_ch=res_ch, ro_ch=cfg["ro_chs"][0])
        gain = cfg["pulse_gain"]

        style=self.cfg["pulse_style"]        
        
        self.add_pulse(ch=self.cfg["res_ch"], name="offset", idata=pulse_I, qdata=pulse_Q)
        self.set_pulse_registers(ch=self.cfg["res_ch"], 
                                 freq=freq,
                                 phase=phase,
                                 gain=self.cfg["pulse_gain"],
                                 style=self.cfg["pulse_style"],
                                 waveform="offset"
                                 )
        
        self.synci(200)

    
    def body(self):
        self.trigger(adcs=self.ro_chs,
                     pins=[0], 
                     adc_trig_offset=self.cfg["adc_trig_offset"])
        self.pulse(ch=self.cfg["res_ch"], t=0)
        self.wait_all()
        self.sync_all(self.us2cycles(self.cfg["relax_delay"]))

config={"res_ch":GEN_CH, # --Fixed
        "ro_chs":[RO_CH], # --Fixed
        "reps":1, # --Fixed
        "relax_delay":1.0, # --us
        "res_phase":0, # --degrees
        "pulse_style": "arb", # --Fixed
        "readout_length":100, # [Clock ticks]
        "pulse_gain":30000, # [DAC units]
        "pulse_freq": 500., # MHz]
        "readout_freq": 500, # MHz
        "adc_trig_offset": 120, # [Clock ticks]
        "soft_avgs":1
       }

prog = IQOffsetProgram(soccfg, config)
iq_list = prog.acquire_decimated(soc)

fADC = soccfg['readouts'][RO_CH]['f_fabric']
fDAC = soccfg['gens'][GEN_CH]['f_dds']
ft_meas = np.abs(np.fft.fftshift(np.fft.fft(iq[0]+1j*iq[1])))
ft_meas /= np.max(ft_meas)
ft_theo = np.abs(np.fft.fftshift(np.fft.fft(pulse_I+1j*pulse_Q)))
ft_theo /= np.max(ft_theo)
f_meas = np.linspace(-fADC/2,fADC/2,len(ft_meas),endpoint=False)
f_theo = np.linspace(-fDAC/2,fDAC/2,len(ft_theo),endpoint=False)
plt.plot(f_meas,np.abs(ft_meas),label="measured by ADC")
plt.plot(f_theo,np.abs(ft_theo),label="sent to DAC")
plt.xlim(-200,200)
plt.xlabel('f (MHz)')
plt.legend();
```

[image]

meeg, 2024-07-10:
Good catch, I think you're correct - the generator and readout seem to have opposite sign conventions, though I think that's not actually the root cause of this behavior.

I haven't dug into it, but my guess is this is sort of as designed, where the generator upconversion and readout downconversion both simply add the given frequency to the data they see:
* you generate a complex waveform, cis(f1*t)
* generator multiplies your complex samples by cis(f2*t), and plays the real part on the DAC - so on the cable you have power at both +(f1+f2) and -(f1+f2)
* readout multiplies the ADC data by cis(f2*t). and gives you the result - so this gives you f1+2*f2 and -f1, and only the latter survives the decimation filter

I suspect that if you use a negative readout frequency you will see the behavior you expect - downconversion should subtract the LO, not add it. Try it?

It should be easy to tweak the API so the readout frequency gets multiplied by -1, but it's not going to be a super high priority to make that change and test it . . . it's pretty rare to downconvert to anything other than DC.

LukeTrainor, 2024-07-10:
That makes sense. I suspect it's probably not worth changing, but definitely worth knowing about.

I think for now I'll just keep using `I-1j*Q`, as I'm ordinarily using a signal generator with a mixer, and it's nice to have the readout frequency be automatically offset by the mixer frequency.

PR #273 (2024-08-03): misc. bugfixes and tweaks
* The register-map fix in #260 didn't get applied to the RF board drivers, and broke them. Fixed.
* NumPy 2.0 changes how type promotion works; this was responsible for #271 and also broke mode code compution in all tProc v1 programs. It looks like now you can run NumPy 2.0 on your PC and still be able to run a QICK board over Pyro, unless more bugs show up.
* Fix #269 in tProc v2 by flipping the sign of the downconversion frequency.
* Some other tweaks to v2 - there's a breaking change, which is that `body()` and `initialize()` are now `_body()` and `_initialize()`.
commit 484594ed (2024-08-03): in v2, flip the sign of the downconversion frequency (fix #269)
closed 2024-08-03 by meeg