openquantumhardware/qick #270: Only 8 out 16 digital pins available.

ibwharri, 2024-07-16:
We have a need to access 14 digital pins, however it seems that QICK only supports 8 out of the 16 digital pins on the ZCU216 we are using. Is this a technical limitation, or is there something we can do to get access to the other 8 pins?

meeg, 2024-07-16:
That's just how many pins are wired in the firmware; you would need to modify+recompile the firmware (starting with the source in https://github.com/openquantumhardware/qick/tree/main/firmware) to get more pins.

Making changes to the firmware is pretty nontrivial and we don't recommend it unless you have some experience - we can give you tips but we can't walk you through the whole learning process.

ibwharri, 2024-07-18:
Would it just be as simple as duplicating these [entries for PMOD0 in the bd file](https://github.com/openquantumhardware/qick/blob/cd0ce4c11c22d073da5753ad1aa1f076f4ebfd30/firmware/bd/bd_216_2022-1.tcl#L2443) to make PMOD1 equivalents (it looks like PMOD1_0 is used for something so maybe that one should be excluded), and then [uncommenting these PMOD1 entries](https://github.com/openquantumhardware/qick/blob/cd0ce4c11c22d073da5753ad1aa1f076f4ebfd30/firmware/xdc/ios_216.xdc#L20) in the xfc file? Or is there something more involved in the firmware that would need to be done?

meeg, 2024-07-19:
I think that's correct, but can't be sure you're not missing something. Normally one doesn't edit the bd.tcl file directly, there's a GUI.

WolfLink, 2024-07-25:
As far as I can tell, from finding posts [like this](https://support.xilinx.com/s/question/0D54U000071I2KgSAK/in-vivado-how-do-i-connect-my-ips-output-port-to-the-leds-on-my-zcu102-board?language=en_US), editing the XDC file directly is the only way to connect more hardware ports.

If there is a way to do it through the GUI please let me know.  I know how to create ports in the block design editor, but I can't find a way to map them to hardware ports through the GUI.

meeg, 2024-07-26:
No, you're right - you can define and wire ports in the block design GUI, but the standard way to map ports to physical pins is using the XDC file e.g. https://docs.amd.com/r/en-US/ug1390-zcu216-eval-bd/Xilinx-Design-Constraints.

meeg, 2025-03-05:
Cleaning up old issues; feel free to reopen if you have some follow-up.

closed 2025-03-05 by meeg