openquantumhardware/qick #274: Uable to run pscp to copy qick tool in RFSoC 4x2 Board

atifjafri, 2024-08-06:
I am trying to copy qick tool in RF SoC 4x2 board using pscp with folloiwng on windows terminal
D:\QICK\cloned_repo>pscp -r qick xilinx@192.168.1.58:/home/xilinx/jupyter_notebooks/. 
but everytime I give the password, it given message of Access denied

[image]

I have copied the pscp.exe in the folder where I have cloned the qick. Before this I tried without copying the pscp.exe (only putty installed), however, with that it was unable to recongnize pscp.

meeg, 2024-08-06:
Are you able to SSH to the board using the same password?

atifjafri, 2024-08-07:
I use factory password of xilinx to access the following
[image]

atifjafri, 2024-08-07:
RF SoC is connected to WLAN router through RJ45  cable and latptop is connected to wifi to access the board as shown in fig. of last message.

meeg, 2024-08-07:
Right, you're showing me the Jupyter server - I was asking whether you were able to connect to the board using SSH (PuTTY, I guess), because that uses the same authentication as pscp. Jupyter and SSH/SCP have the same default passwords but work in different ways.

If you can't get pscp to work, you could zip the QICK repository, upload the .zip file through the Jupyter server, then open a terminal in Jupyter and use that to unzip the .zip file.

atifjafri, 2024-08-07:
The solution you proposed is working. Thanks

closed 2024-08-07 by atifjafri