openquantumhardware/qick #288: Can't read 4x2 metadata with 0.2.287

i-am-mounce, 2024-09-26:
I'm stuck in a loop updating QICK-DAWG with some compatibility issues:

1. With version 0.2.160 I'm having issues with Pyro and somehow sending over np.int64 and float64's being an issue.  I think this is fixed in some later version of QICK. 
2. With 0.2.287 I'm getting an error that QickSoc() cannot even read the as provided RFSoC4x2 metadata.

meeg, 2024-09-26:
1. Yes, I think #273 fixed this (though the fix is pretty ugly).
2. Will check on this. Are you seeing this with your firmware that has axis_avg_buffer:1.1, or is it also happening with the standard 4x2 firmware?

i-am-mounce, 2024-09-27:
Seems standard firmware. I actually didn’t have an issue with axisaveragebuffer1.1

meeg, 2024-09-27:
Can't reproduce on 0.2.287 and both standard and buf1.1 4x2 firmwares - I tried both with and without Pyro. This fails for you, right?
```
from qick import *
soc = QickSoc()
# soc = QickSoc("/home/xilinx/jupyter_notebooks/fw/2024-06-14_4x2_commonclk-4pt9/qick_4x2.bit")
soccfg = soc
print(soccfg)
```

i-am-mounce, 2024-09-27:
Yes. That and default firmware. Let me do a fresh pynq image and try again. Maybe I screwed something up

i-am-mounce, 2024-09-27:
Ok, fresh pynq image and qick install worked. Upgrading to 0.2.287 now, then i'll make a pull request for the photon counting!

closed 2024-09-27 by i-am-mounce