openquantumhardware/qick #290: Custom register doesn't like int32

i-am-mounce, 2024-10-08:
I'm trying to run a microwave length sweep with a custom register (I don't know if you guys update this to work more easily). But with the most recent version I'm getting an error:

[1222](file:///C:/snl/qick-dawg/.venv/Lib/site-packages/qick/asm_v1.py:1222) # print(inst)
   [1223](file:///C:/snl/qick-dawg/.venv/Lib/site-packages/qick/asm_v1.py:1223) for field in fmt:
-> [1224](file:///C:/snl/qick-dawg/.venv/Lib/site-packages/qick/asm_v1.py:1224)     mcode |= (args[field[0]] << field[1])
   [1226](file:///C:/snl/qick-dawg/.venv/Lib/site-packages/qick/asm_v1.py:1226) if inst['name'] == 'loopnz':
   [1227](file:///C:/snl/qick-dawg/.venv/Lib/site-packages/qick/asm_v1.py:1227)     mcode |= (0b1000 << 46)

OverflowError: Python int too large to convert to C long

This seems to be because the register is trying to be set to (25 <<56) | np.int32(4) which overflows the new int32.  The register value is being converted at the QickRegister.val2reg function as the final else statement:

    def val2reg(self, val):
        """
        convert physical value to a qick register value
        :param val:
        :return:
        """
        if self.reg_type == "freq":
            return self.prog.freq2reg(val, self.gen_ch, self.ro_ch)
        elif self.reg_type == "time":
            if self.gen_ch is not None:
                return self.prog.us2cycles(val, self.gen_ch)
            else:
                return self.prog.us2cycles(val, self.gen_ch, self.ro_ch)
        elif self.reg_type == "phase":
            return self.prog.deg2reg(val, self.gen_ch)
        elif self.reg_type == "adc_freq":
            return self.prog.freq2reg_adc(val, self.ro_ch, self.gen_ch)
        else:
            return np.int32(val)

Am I using this wrong or is this a bug? It didn't seem to be an issue with 0.2.186.

meeg, 2024-10-08:
Are you maybe running with numpy 2.0? We've been running into problems with how 2.0 handles type promotion, e.g. #273 and this looks like more of that.

The laziest possible workaround for that would be to replace the last line of val2reg with `return int(np.int32(val))`, maybe that will work?

i-am-mounce, 2024-10-09:
Yes, it's numpy 2.1.1. and the fix works.  I guess the mcode is 64 bit but we want to restrict the register value to 32 bit the lazy version's not bad. Should I do a pull request?

meeg, 2024-10-09:
Great - sure, if you do a PR that'll be great, otherwise I'll get to it this week when I'm ready to merge some other stuff in.

meeg, 2024-10-10:
I got to it, #291.

closed 2024-10-10 by meeg