openquantumhardware/qick #292: Help with Generating Multi-Tone Signals on RFSoC 4x2 with QICK

ljr145733, 2024-10-12:
Hello,

We're working with an RFSoC 4x2 board and are exploring its potential use for controlling a qubit laser’s AOM. We're particularly interested in generating multi-tone signals and were wondering if this can be easily achieved using QICK.

We've noticed the muxed generator feature but aren't sure how to utilize it for our application. Could you provide any examples or guidance on this?

Any help would be greatly appreciated.

Thank you!

meeg, 2024-10-12:
The available 4x2 firmwares don't have muxed generators, so you would need to build your own firmware.

You can find the sources for the 4x2 firmware project and the muxed generator, and some brief instructions on how to generate the project, in https://github.com/openquantumhardware/qick/tree/main/firmware, but we don't recommend attempting this unless you have some experience with FPGA design and the Vivado tools.

ljr145733, 2024-10-12:
Thank you so much for the fast reply. 

I guess we will try to use IQ modulation then although it might be slower.

closed 2024-10-13 by ljr145733